bitdump/BLHeli #471: Windows Defender detecting Trojan:Win32/Spursint.F!cl

luke5500, 2020-07-16:
Hello, I am trying to download the BLHeli_32 zip but Windows Defender is blocking it stating it has the following virus detected Trojan:Win32/Spursint.F!cl. Is this a known false positive? Thanks

4712, 2020-07-16:
@luke5500 Whats exactly the file name of the zip you try to download and from where?

luke5500, 2020-07-16:
Yep, those are the links I am trying and both give the same virus detected warning. I am downloading the BLHeliSuite_32 zip for Windows.
[image]

4712, 2020-07-16:
I cannot reproduce it here with Win10 2004 defender.
I checked the files again and found they are binary identical, so it is a false positive warning.

luke5500, 2020-07-16:
Ok, thank you for checking. I just wanted to make sure.

closed 2020-07-16 by luke5500

4712, 2020-07-16:
Very understandable, I usually use [virustotal](https://www.virustotal.com) to check the files. 
SecureAge APEX  detects it "Malicious". Reported False Positive again here...