bitdump/BLHeli #472: ***INVALID*** ESC Warning - Cannot Flash - MacOS

jameskmulligan, 2020-07-16:
Hi Team,

New FPV addict here. Having an issue with BLHeli and can't seem to find a remedy online.

All motors work in Betaflight - but two were going the wrong direction. Opened up BLHeliSuite32xm. On MacOS - El Capitan 10.11.6. Managed to connect to the drone, but when i read setup, one of the ESCs is reading as INVALID. See error:

[image]

I click OK, and it shows me this prompt:

[image]

But when i hit 'Yes' i get this error message:

[image]

What should i do to fix this? I'm really hoping it's a problem with my Mac, and if i can fix it, i'll be able to fix the ESC. So very close to my first flight!

Cheers,
James

haslinghuis, 2020-07-17:
Related / Duplicate of #324 

Please check if updating OpenSSL does solve this: For more details see the issue linked.

4712, 2020-08-26:
@jameskmulligan added support for MacOS El Capitan and Sierra in BLHeliSuite32xm 1.0.0.5. Please check and close this issue.