bitdump/BLHeli #481: BLHeli32 DShot "Motor Direction" configuration different than PPM

josephduchesne, 2020-09-10:
I'm using a BLHeli32 motor controller with a PID loop, and recently switched between PWM input (Motor direction: bidirectional soft), and dshot600 (sending command ID 10 to enable bidirectional input on the ranges 48-1047, and 1048-2047.

I was expecting to be able to run a faster PID loop, but what I found is that the PID controller's stability is a lot worse. I think that this might be because in dshot, I'm not able to set "bidirectional soft", only "bidirectional 3d". Is there any way to control "bidirectional soft" using dshot?

sskaug, 2020-09-11:
The Dshot command can not select soft bidir mode. Only 3D bidir mode. So you will need to use BLHeliSuite32 or the BLHeli_32 app to configure soft bidir mode. Then pwm and Dshot inputs behave similar, except that for pwm, there is a deadband, which is not needed with Dshot (since Dshot is an accurate digital value).

josephduchesne, 2020-09-11:
So for clarity, even if I set soft bidirectional mode in BLHeliSuite32, DShot won't use it?

sskaug, 2020-09-11:
If you select soft bidir in BLHeliSuite32, bidir will be soft.
But if you then send bidir Dshot command, bidir will be 3D.

josephduchesne, 2020-09-12:
Will the bidirectional control be soft with dshot if I reverse direction using the direction1/2 commands?

sskaug, 2020-09-13:
Yes, it will. Both for commands 7 and 8, as well as for 20 and 21.