bitdump/BLHeli #482: Found no valid ESC configuration on connection to pixhawk 2.4.8

prem48science, 2020-09-19:
Hi, I am trying to connect and calibrate ESC through pixhawk. and blheli32 cant recognize or read data of ESC through pixhawk. it throws error found no valid esc configuration. please suggest a solution

4712, 2020-09-19:
@prem48science Maybe this helps: https://github.com/bitdump/BLHeli/issues/391

4712, 2020-10-12:
@bestaps if you have still problems with the connection, you might want to post the *.param file here.

bestaps, 2020-10-13:
@4712  the error got resolved.

mantisboxer, 2021-02-01:
I'm having this problem as well on my Pixhawk Mini. I have attached my parameters file from Mission Planner. BLHeli32 Suite connects to the FC no problem, but doesn't seem to recognize the ESCs.
[MCD_2021_Q4X.zip](https://github.com/bitdump/BLHeli/files/5902449/MCD_2021_Q4X.zip)

4712, 2021-02-03:
@mantisboxer I do not know much about the Pixhawk Mini, but if you connect the the FC using the USB port you need to set SERVO_BLH_PORT=0, AFAIK.
And disconnect MissionPlaner, while using BLHeliSuite32...