bitdump/BLHeli #484: Impossible to flash 2 of the 4 esc (mobile 6)

seko4, 2020-10-25:
Hi,

I can not flash ESC 1 and ESC 2, but it is ok for ECS 3 and ESC 4
There is a strange name for the firmwares of ESC 1 and ESC 2
And the "Flash All" button is not activated.

Any clue about that ?

below my debug file and a snapshot of my screen

[image]

[Log-BLHeli.txt](https://github.com/bitdump/BLHeli/files/5434331/Log-BLHeli.txt)

mikeller, 2020-10-25:
@seko4: Betaflight 3.5.7 is a very old and unsupported version. Please update the flight controller firmware to a recent version and try again.

seko4, 2020-10-25:
I was on betaflight 4.2.4...

I made it once more to be sure,
and then I retried to flash ESC1.
I can select the file to flash, press the "flash" button,
but nothing happens...

Here is the log file for this try...
[newlog.txt](https://github.com/bitdump/BLHeli/files/5434952/newlog.txt)

[image]

mikeller, 2020-10-25:
Looks like 2 of your ESCs are not properly responding. Do you have a battery plugged in when attempting to flash (props off)?
Also, can you please post the output of a `diff` in Betaflight configurator CLI?

seko4, 2020-10-26:
Here is my diff.

Is there a way to reset the 2 ESC that does not respond properly ?

[BTFL_cli_MOBULA_6_20201026_161133.txt](https://github.com/bitdump/BLHeli/files/5439618/BTFL_cli_MOBULA_6_20201026_161133.txt)

mikeller, 2020-10-26:
@seko4: The diff looks good - I think it's the ESCs that are stuffed. for some ESCs it is possible to connect an ICSP header for each ESC and re-write the boot loader, but I suspect that the CRAZYBEE is too small for this.

seko4, 2020-10-26:
I managed to flash the 2 problematic esc, by choosing the option "Ignore inappropriate MCU and layout", and it worked !

closed 2020-10-26 by seko4