bitdump/BLHeli #486: Tekko32 F3 4in1 ESC 45A (Revision C) - electrical noise on video

maxgame91, 2020-10-31:
Hello,

I have a this [stack](https://www.drone-fpv-racer.com/kakute-f7-v15-tekko32-f3-4-in-1-esc-3773.html), but when I fly I have massive noise on the [video](https://www.youtube.com/watch?v=x0L5oTBNswg&feature=youtu.be) (here a test without props, only pitch and roll movement)

So I read this [post](https://github.com/bitdump/BLHeli/issues/468) and update my esc with blheli 32.7.2 and still have the same problem.

I tried many vtx, camera, frame, but nothing change. I tried 48khz and 96khz but it's still the same
I have a 1000uf 35v low esr directly solder to the xt60. camera and vtx are also on the xt60.
And tried 4s and 6s
Anyone have a solution with that ESC ? saw many problems

Best regards

theely, 2020-10-31:
Hi,
I have reported a similar issue #485, electrical noise is one of the symptoms however the main issue is the inability of the ESC to properly ramp up the motors. It might be possible that this issue is due to a production defect.

Would you have the possibility to take a DSHOT_RPM_Telemetry log and do a couple of half flips and rolls?

maxgame91, 2020-11-01:
Hi
@theely I saw your report, thanks for details,
I will rebuilt the drone, and do some flips and rolls

Keep you in touch