bitdump/BLHeli #487: [Feature Request] Stop/Break motor at fix position

the-nicolas, 2020-11-01:
For many VTOL UAVs it would be great (or even necessary) to stop the rotors at fix "parking" positions. Sometimes just to reduce drag or because they get retracted.

Right now the most common approach seems to be:
- add some reflective point to one (or more) rotos blades (if blade is not already white) 
- install some IR Led/Sensor on motor arm (sensors are available). 
- have some own electronics which gets activated when rotors are not in use which is attached in parallel to motors to rotate some slowly to position and apply some break current. 

The feature is about:
- have some input for the IR sensor
- break motor
- slowly rotate motor till sensor gets activated
- hold break

What do you think about it?

sskaug, 2020-11-03:
Sounds like a feature that can be implemeted in an ESC.
But it is a bit outside of the main market of BLHeli, and as such I believe it adds too much complexity to support a function that only a very small portion of the market will use.

the-nicolas, 2020-11-03:
I had the same thoughts, but I think the rising fpv wing community is also interested in such a feature. They also have the problem with breaking propellers during landing. The folding props are way more expensive and there are way less products available - and they have also other disadvantages.

I think that feature would increase the adaption of blheli esc's in the fixed wing market. The reason why today people use it for fix wings is mostly to possibility to reverse the thrust - which is not supported by most normal Esc's. And reverse thrust is a nice thing for auto landing wings by FC to reduce airspeed during landing (... and potential damages).