bitdump/BLHeli #488: [Feature request] Support 24 kHz PWM frequency on Tekko32_F3_C

SJChannel, 2020-11-02:
The Holybro Tekko32 F3 40A 4-in-1 ESC (Tekko32_F3_C firmware) ships with the PWM frequency set to 96 kHz, and the lowest it can be set is 48 kHz.  Performance is terrible at 96 kHz, with very low authority at low throttle leading to large bobbles on throttle cuts.  At 48 kHz, the performance is tolerable but still not very good.  I am requesting the ability to reduce the PWM frequency farther to 24 kHz, if possible.

I realize that the problems at 48 kHz can be minimized with careful tuning in Betaflight.  However, all of my 24 kHz ESCs perform well out-of-the-box on the Betaflight default tune.  The "smoothness" that is claimed for higher PWM frequencies is of no value if the quad cannot track the sticks properly.

sskaug, 2020-11-03:
BLHeli_32 can support various pwm frequencies, and as pwm frequency relates to hardware design, the choice of the range of pwm frequencies is up to the harware manufacturer. And in order to ensure that this range is abided to even in fault situations, the range is coded into the bootloader. So you will need to bring this issue to Tekko.

What we can do for a test code, is to generate a code that always runs at half the programmed pwm frequency.
Code is now here: https://github.com/bitdump/BLHeli/tree/master/BLHeli_32%20ARM/Half%20pwm%20frequency%20test%20code

SJChannel, 2020-11-03:
Thank you very much for generating the special test code!  I'm looking forward to trying it using your variable PWM feature over the range 24-48 kHz.  At the moment, the weather outlook is very poor here, but I'll report back after I've been able to test it.  I will also open a support ticket with Holybro on this topic.

SJChannel, 2020-11-14:
I flew a few packs using your test code and a 24 kHz PWM frequency.  It works great.  Thanks again!

closed 2020-11-14 by SJChannel

cobonovi, 2021-12-20:
Hey guys, I flashed the firmware but it still shows 48khz PWM, should it be like that?

SJChannel, 2021-12-21:
Yes.  The actual PWM frequency is half of the value selected in the configurator.