bitdump/BLHeli #489: Motors blowing when plugging in battery

FlexFreak, 2020-11-03:
We enabled bidirectional in BLHeli-Configurator and wrote the setup. Then we plugged in the battery and the motors tryed to spin, got hot and then they blew up. After disableling this setting the motors stopped trying to spin. We didn't even turn on the motors it just happend when we plugged in the battery. Before we enabled this feature the motors and the esc worked fine. So now on three out of four motors the windings are fused.

mikeller, 2020-11-04:
@FlexFreak: This GitHub project is for the BLHeli firmware and the BLHeli Suite, but not for [blheli-configurator](https://github.com/blheli-configurator/blheli-configurator).
Can you please provide some more information:
- what is your hardware configuration?
- what firmware is running on the flight controller that your ESC is connected to, and what configuration (especially motor output protocol / configuration) did you use?
- was 3D mode enabled in the flight controller firmware?

FlexFreak, 2020-11-05:
I'm using a MAMBA F405 Mini Stack with Betaflight 4.1 and Dshot600 and 3d mode wasn't enabled.

mikeller, 2020-11-07:
@FlexFreak:
> We enabled bidirectional in BLHeli-Configurator and wrote the setup

> I'm using a MAMBA F405 Mini Stack with Betaflight 4.1 and Dshot600 and 3d mode wasn't enabled.

ESCs in bidirectional mode without 3D mode enabled on the flight controller is not supposed to work. It _should_ not result in fried motors, but then it's also not massively tested as it is not a supported combination. Please read some tutorials / watch some instructional videos before attempting to configure a craft.