bitdump/BLHeli #492: Tekko32 F3 45A Revision C - Desync under heavy load

theely, 2020-11-13:
Hi there,
could you help me debug an issue that is affecting some batches of Tekko32 F3? Looking for advices on how to further test and isolate the issue.
I would like to understand if the ESC is defective or if it requires different BLHeli settings to run property.

**Expected behaviour:** On half flips and rolls the motors should properly ramp up without de-sync or delay. This is the case with  Tekko32 F3 45A with serial number: 31023 004 200829, the log bellow illustrates how motor 2 and 1 properly ramp up to end the roll.
[image]

**Observed issue:** When replacing the working ESC with a newer batch, serial number: 31023 005 200928 (I have tried two different ESCs with the same serial number). The ESC struggles to ramp up properly the motor and in some occasion goes in desync.
[image]
[image]

How was the test performed. I have swapped a properly working  Tekko32 F3 with a problematic one without changing any other variable. De facto isolating the problem to the ESC.
Next I have tested several configurations, some mitigated the issues others made things worst. See table below:

Change | Results
------------ | -------------
Installed vanilla BF 4.2 on default settings | No difference, issue still present
Disabled Bidirectional DSHOT | No difference, issue still present
Re-flashed BLHeli | No difference, issue still present
Reduced PWM from 96kHz to 48kHz | Got worst, immediate desync on flips and rolls
Reduced ramp up power to 25% | Small improvement less de syncs
Motor timing to 26 | Small improvement less de syncs
Disabled low RPM protect | Improvement but ramp up takes much longer then expected

Re-flashed test BLHeli 32.7.3 | Got worst, immediate desync on flips and rolls
https://drive.google.com/drive/folders/11mEeKe6_ok3_siuswv1yySXTsgk_OoO7?usp=sharing

Please find here all log files of the different tests:

Xedos9er, 2021-02-01:
Glad to read this here. I had the same issue. Only happening on fixed 48khz pwm. I had similar problems on my old Tekko32 F3 40A ESC when running 48khz. Where can I check the serial number.

theely, 2021-02-01:
The serial number can be found on a sticker on the ESC packaging box.

theely, 2021-02-01:
Status update: 
Holybro told me that the ESC with serial number 31023 004 200829 was version 4 and the ones with serial number 31023 005 200928 were version 5. There is only a marginal difference between the two versions.

After sending in the defected ESCs I received a working replacement. Unfortunately I was not provided with any information about the reasons for the ESCs to not perform as expect.

Xedos9er, 2021-02-01:
Looks like those Tekko32 F3 ESC´s are a huge mess. Never had issues with other brands. 
I will get in touch with holybro.