bitdump/BLHeli #494: dshot rpm data becomes more noisy when we increase motors' speed (

ouafi98, 2020-11-23:
[image]
Why dshot rpm data become noisier when we increase motors' speed?
I'm using SucceX-A F4 AIO board and my ESC's firmware is iFlight-BL32-4IN.

sskaug, 2020-11-23:
The erpm data from BLHeli is not filtered in any way, and is quite noisy, yes. An important reason for the jitter, is the mechanism used to avoid having nonlinearities where the harmonics of commutation frequency matches the pwm frequency.

ouafi98, 2020-11-23:
if possible, How can I fix it by myself? 

Thank you for you quickly reply. I don't understand this "An important reason for the jitter, is the mechanism used to avoid having nonlinearities where the harmonics of commutation frequency matches the pwm frequency." (can you give more references?)
On KISS' esc I didn't see this behavior and I try to figure out what's going on because I want to use BLHELI to develop force control systems.

sskaug, 2020-11-24:
This is the way BLHeli_32 delivers data - totally unfiltered so that users can apply filtering and have the flexibility to trade off filtering versus delay. If you can afford some latency, you can of course low pass filter the data to reduce noise.

closed 2020-11-30 by ouafi98