bitdump/BLHeli #495: HGLRC Forward 60A BLHeli32 - 32.7 Overheating

emolina82, 2020-11-26:
I'm having heating problems in my ESC with version 32.73: HGLRC_Forward_ESC_Multi_32_73.Hex

Looks like everything works fine in 32.6 version, but something goes wrong in 32.73 (and al 32.7? that I've tried)

Is it normal to have decompensation motors when I test the motor in BLHeli32 suite or betaflight. When I start to spin the motors .. they work on their owns and aggressive movement more than 32.6 version. The motors are warmed (even hot) after betaflight tests without props (supposedly in this tests there is not gyro interaction), is it normal?

I've tested hover my room with this configuration and before a minute I should turn off the ESC because of the extreme heat

Could you help me?

mikeller, 2020-11-26:
@emolina82: Have you tried the 32.71 version that can be installed in [BLHeli Configurator](https://github.com/blheli-configurator/blheli-configurator/releases)?
Unfortunately all other 32.7X 'releases' are a complete mess of unfinished and not fully working features, often released under the same version number, so it is impossible to tell what will work and what will burn your ESCs / motors.

sskaug, 2020-11-26:
@mikeller, what are you commenting on here? BLHeli_32 can not be flashed with BLHeli Configurator.
All the 32.7.x codes are well controlled and with tracable changes. 
And that they burn ESCs or motors is a new assertion.

mikeller, 2020-11-26:
@sskaug: Sorry, my bad, I confused them with the inofficial versions of BLHeli_S. Removed.

sskaug, 2020-11-26:
@emolina82 One error that can cause motor overheating is motor screws extending into windings. Have you checked this?

emolina82, 2020-11-26:
Hi @sskaug 

I already checked this and it's ok because I used the short screws that came with the motors

I also checked if there is electric continuity between screws and ESC soldering and there isn't 

I've tried many BLHeli 32 version, from 32.66 to 32.73 and all of them works wrong, the motors spin at different times. Te only version that looks like is working well is 32.6 but I can't activate bidirectional communications in betaflight (despite version 32.6 looks like is working well, the motor get warm without propellers too, could be a physical problem with the ESC)

I've changed the order of the motors in betaflight (resource remapping) because my ESCs are not alienated with the motors in the frame.

Many thanks for your help

sskaug, 2020-11-27:
There are tons of ESCs out there running 32.7 and later test codes without problems, so something is special or defective in your setup. What about 32.7 without bidirectional dshot, does that give you problems too?

emolina82, 2020-11-27:
thanks @sskaug 

yes, I've tried without bidirectional dshot and still works wrong

First of all, I would like to know if it is normal under 32.7 firmware that the motors spin at different times in betaflight motor tab when I increase throttle slowly, Is it happen to everybody under this version? Is it a correct behaviour or is a problem with my ESC or my setup?

What I think know is there is a problem with ESC version 32.7 for my ESC model. I mean, there is another HGLRC Forward ESC, but mine with 60A is almost new, probably others version from HGLRC Forward doesn't work well with HGLRC Forward 60A, could it be possible?

If I fly on 6s (hover) the ESC starts to gain temperature and I should stop (my first flight on 6s my quad fallen in the country)

Yesterday I was hovering my room and my quad move to the left out of the blue (I didn't touch the sticks) I think that was a problem with my capacitor (the soldering was broken)

Regards