bitdump/BLHeli #497: 32.73 - short motor stutter

TheIsotopes, 2020-11-26:
I have tested 32.73 beta on Aikon_AK32_35A_Multi and Aikon_AK32_4S_V1_0_Multi and i am not satisfied.

On both versions, with fast throttle cut, I noticed that the engines fail for a short time. feels like a stutter or a briefly swallowing of the motors. this is particularly noticeable with fast rolls and flips. im using default settings with same pwm frequency on low and high throttle. with variable frequency the problem gets worse.

Going back to 32.72 and the problem is fixed.

sskaug, 2020-11-26:
So you mean that when you quickly drop throttle, the motors momentarily stutter/fail to produce thrust?
I tried throttle steps (fast up/down) on the bench, but can not notice any misbehaviour.

Any more info you have - what is your setup, log files etc?

TheIsotopes, 2020-11-26:
> So you mean that when you quickly drop throttle, the motors momentarily stutter/fail to produce thrust?

Yes, maybe it could also be a motor desync. this occurs more often on my F4 system than on my F7 system.

> Any more info you have - what is your setup, log files etc?

Unfortunately, I don't have any log files. But i can test and log this again later on my F4 system.

The Aikon_AK32_35A_6S working with a Matek F722 Mini and latest BF 4.3 (8k/8k,DShot600,bidir)
and the Aikon_AK32_35A_4S are using with a Matek F405 Mini also on latest BF 4.3 (8k/4k,DShot300,bidir)

TheIsotopes, 2020-11-29:
@sskaug I'm a little speechless now.
was flying with the f4 today, current BF4.3 and esc with 32.7.3, flies absolutely cleanly, could not detect any motor misfires. The only difference to the last flights is that the bf now have `Dynamic idle and idle related mixer changes`. But I don't use it.

I'll close here now, should the problem arise again, I could open it up again.

sskaug, 2020-11-29:
Great - good to hear!
But don't hesitate to open again.

TheIsotopes, 2020-12-02:
@sskaug ok, open again.

flew again today and I noticed another loss of thrust wiht my F4 / Aikon_AK32_35A_4S under full throttle.
Here the log file and a sreenshot from loss of thrust:
[btfl_004.txt](https://github.com/bitdump/BLHeli/files/5630055/btfl_004.txt)

[image]

sskaug, 2020-12-02:
Thank you for the logfile. I will look into it.....

TheIsotopes, 2020-12-14:
@sskaug is there any news?
I was flying again today, dropouts on all three systems with f4, f7 and h7 mcu.
Pls can you create `Aikon_AK32_35A_Multi` and` Aikon_AK32_4S_V1_0_Multi` from your current build without sbus and variable pwm frequency support for testing. I'm of the opinion that the dropouts could be related to the variable PWM frequency.
Therefore I would like to test the esc firmware without additional support for other functions.

sskaug, 2020-12-15:
I did study the log, and I did try to recreate such a scenario on the bench several times, but was not able to see something that could resemble stutter. It's also puzzling that there should now probably be more than 100 uads out there with the 32.7.3 code and  I have not seen any problem reports relating to this.

Great that you are willing to test! There are quite a few changes between 32.7.2 and 32.7.3, I have taken some of the more significant ones and grouped them to generate 4 testcodes with a different set of changes. 
https://github.com/bitdump/BLHeli/tree/master/BLHeli_32%20ARM/Short%20motor%20stutter%20testcode
Whatever you can test will be appreciated.

sskaug, 2020-12-15:
Btw, when you experience these stutters, is it always the same scenario as the log? All motors operating near full throttle?

TheIsotopes, 2020-12-15:
> Whatever you can test will be appreciated.

This can take a while, but i do my best to test everything.

> Btw, when you experience these stutters, is it always the same scenario as the log? All motors operating near full throttle?

Yes, it looks like it always happens at almost full throttle.

[image]
[image]

Asizon, 2020-12-18:
Betaflight 4.3 is not an stable version to make test with Blheli32 beta firmwares, it can create a confusions. I think is better to try with an stable 4.2 version to make sure that the problem is not in Betaflight side.

TheIsotopes, 2020-12-19:
@sskaug Here is a status update of the test series.

Test systems:
1) Matek F405 Mini with Aikon AK32 4S and EMAX ECO 2306 2400KV (8K/4K,DShot300,bidir)
2) Matek F722 Mini with Aikon AK32 6S and EMAX ECO 2207 2400KV (8K/8K,DShot600,bidir)
3) SPRacing H7 Nano with Aikon AK 32 6S and EMAX ECO 2207 2400KV (8K/8K,DShot600,bidir)

all systems work with bf 4.3.0 current master and were flown with 2x3S and 2x4S batteries (i.e. four flights per system).

Test results:
- Test code No1 and No2 worked on all systems with no apparent problems.
(i.e. nothing heard during flight / nothing found in the log file)

- Test code No3 worked without problems on System1 and System2.
On System3 the first two flights (3S battery) had no problems,
the third flight had two dropouts and the fourth flight had one dropout, both with 4S battery.

- Test code No4 has not yet been tested.

TheIsotopes, 2020-12-26:
@sskaug As you can see above, I have completed all tests.
Since I don't know what you changed in the test codes, it would be nice to know if you could identify the problem.

Honestly i have no plan what will trigger this problem.
The only clue is that this always occurs at full throttle, otherwise I can't see any other pattern.

sskaug, 2020-12-27:
Thank you for the extensive testing!
So for the test codes, there were really only some dropouts on test code 3.
I'm not sure it is possible to draw any clear conclusions from this.

The test codes were built like this:
[image]
The 8 rows represent the mods/fixes done to the test code after Rev32.7.2.
An X in a cell means that that mod/fix was not implemented in the test code.

So the only code that had dropouts was the code that did not have variable pwm implemented.

Can you make any sense of it?

TheIsotopes, 2020-12-27:
At first I thought that it would be due to the variable PWM frequency.
I didn't think that so many changes compared to 32.72 have been made.
I have absolutely no idea, it always seems very sporadic to me (today it's good and tomorrow it's back).
Unfortunately, this problem has not yet been reported by anyone else, which makes me very thoughtful.

TheIsotopes, 2021-01-02:
@sskaug I have now looked at everything again and I came to the decision to give Test code No4 another chance.

Why? this code has implemented most of the fixes and there have been no dropouts so far.
Even after today's flying this code is my favorite. As in the last test, there are no dropouts today on any of the three systems.

The question arises whether `Input SD` and `Full power step` in combination causes the dropouts.
Separated as in Test code No1 or Test code No2, there were no dropouts. They were included in Test code No3 and there were dropouts.

What do you think? could it have anything to do with it?

sskaug, 2021-01-03:
Great that you continue testing. I would not say that the input Sigma Delta modulation in itself is likely to cause dropouts, but the full power step change is more risky, as it interacts directly with hardware. I will continue to look into it - great if you collect more flying time to gain more confidence in code 4 hot having the issue.

sskaug, 2021-01-04:
So I have now gone through the recent changes in detail. And found that the full power step mod can cause issues. Also it does not properly do what it was designed for. So that mod is now removed from the new Rev32.7.4 test code that is now published.
The variable pwm mod is also improved, it could in some situations cause deadtime to be shorter than intended.

sskaug, 2021-01-10:
@TheIsotopes, have you had a chance to test Rev32.7.4?

TheIsotopes, 2021-01-10:
So far the weather has always been bad, but i hope i can test something today.

TheIsotopes, 2021-01-10:
@sskaug I flew five batteries each with two of my systems. Could not find a dropout either while flying or in the logs.
It currently looks good, but I will test a little more to see if it stays that way.

imo you could delete the folder with the test codes on github, these are no longer needed.

sskaug, 2021-01-10:
Great, thank you very much for the help!
Closing this for now, then. Please reopen if you later still see issues.

closed 2021-01-10 by sskaug