bitdump/BLHeli #498: Arduino Uno - ESC not found

tguillaume02, 2020-12-02:
Hello,
It's my first time with BLHeli.
I have bougth an ESC and I would like read information.
I use and arduino UNO, the esc is connected to GND and D11 on Arduino.
on This video we can see of BLHeli don't found my ESC
 [https://youtu.be/mmlfpy6GHU8](https://youtu.be/mmlfpy6GHU8)

The connections:
[http://prntscr.com/vui693](http://prntscr.com/vui693)

The ESC : 
[http://www.readytosky.com/e_productshow/?1222-30A-Brushless-Motor-Speed-Controller-Control-RC-BEC-ESC-1222.html](http://www.readytosky.com/e_productshow/?1222-30A-Brushless-Motor-Speed-Controller-Control-RC-BEC-ESC-1222.html)

Could you help me to find the reason please ?

4712, 2020-12-09:
ESC does not run BLHeli nor BLHeli_32, so it cannot be found. (see https://github.com/bitdump/BLHeli/issues/499)

tguillaume02, 2020-12-10:
it's maybe a N76E003AT20

IAMHAADICOOL, 2023-10-12:
> ESC does not run BLHeli nor BLHeli_32, so it cannot be found. (see #499)

Can you please explain a little bit more? I have the following ESC and I am encountering the same error
https://robu.in/product/emax-blheli-series-30a-esc-oneshot-original/

4712, 2023-10-12:
This is definitely not BLHeli_32, and may be not even connectable using BLHeliSuite. [Here ](https://emaxmodel.com/products/blheli-series-30a-esc-oneshot-available#)is stated "_Based_ on BLHeli"...

IAMHAADICOOL, 2023-10-12:
> Is this the right one?

Can you please suggest to me an ESC that is BLHeli_32 and supports
bi-directional rotation, basically that can be used with Blue Robotics T200
thrusters. Please can you help?
On this website?
robu.in
At lowest price and most suitable?
ᐧ

IAMHAADICOOL, 2023-10-13:
Hey buddy, can you please reply? I really need all the help I can get. I
have to submit this project and I really hope that you could help me out.

ᐧ