bitdump/BLHeli #502: Set ESC in bidirectional mode from my mcu

lyris85, 2020-12-10:
Hello,
I have purchased an ESC with BLHELI32 firmware. I want to configure its firmware to be in bidirectional mode. I have my owm microcontroller and I would like to connect it to the ESC to configure it. What command(s) and protocol should I use to set it in bidirectional mode?
Thanks

mikeller, 2020-12-11:
@lyris85: You can use Dshot commands to achieve this - have a look at https://github.com/betaflight/betaflight/blob/57c9c127810b3886273f56a5d90d3022cd6ca5e4/src/main/drivers/dshot_command.h#L44 and related files.

lyris85, 2020-12-17:
Thanks, it works.
I can now use it in bidirectional mode. However, every 1s or so the motor make a spike while running. Like it misses a pwm frame or so. I use a simple PMW frequency of 50hz for testing. I did not have this behavior with my RC car ESC using the same code.

I'll try to set the frequency at 24khz to try to simulate dshot.
Any idea if it comes from the pwm frequency ?

lyris85, 2020-12-23:
The problem was the "Ramp up power" setting by default at 50%. I set it at 100% and everything works like a charm.

closed 2020-12-23 by lyris85