bitdump/BLHeli #505: Dead time configuration

thestafman, 2021-01-05:
Hello. I am making my own ESC based on BlHeli_S design with super fast FETs and I want to get the dead time adjusted to only one time step instead of the minimum 5. I noticed that the dead time steps are in jumps of 5 clock units. Is this just a matter of convenience? Nothing wrong with going 1 or 6, for example? Also. How would I be able to change it? Do I just find and replace the 5 steps with one in the .inc file? Thank you

sskaug, 2021-01-05:
Anything from 1 and up is ok. Zero is dedicated to EN/PWM style drivers.
Just change the "FETON_DELAY EQU XX" definition in the asm file, and the according "ELSEIF FETON_DELAY == XX" in the inc file.

thestafman, 2021-01-06:
Thank you @sskaug for your prompt reply.

thestafman, 2021-01-06:
One more question - How do I generate a .hex file from .inc files?

thestafman, 2021-01-06:
@sskaug I found the makehexfile.bat file and I navigated the .inc files and found where I would change the dead-time (as well as pin configuration). However when I searched for "FETON_DELAY EQU" in the .asm file

;Select the fet deadtime (or unselect for use with external batch compile file)
;FETON_DELAY EQU 15	; 20.4ns per step

Now I am confused about what an external batch compile file is. Should I unhash this comment ? or should I just play with the .inc file? Also, if you can give me an idea of how the .bat file works that would be really appreciated.

sskaug, 2021-01-06:
You'll have to compile the code. This can be a starting point to learn it: https://github.com/bitdump/BLHeli/issues/226