bitdump/BLHeli #506: BLHeli_S support on BLHeli ESCs

benno1237, 2021-01-07:
I know its not really an issue but i would really love to get DSHOT support for my BLHeli ESCs (EMAX Lightning 20A).
On Discord i got told that flashing BLHeli ESCs to BLHeli_S actually is possible if only one would know the configuration. Could you help me out? The ESCs use 7418 FETs as far as i can tell.
I would really appreciate that!

yes i accidently posted this in the betaflight/blheli_s repo before, sorry

Krilatii, 2021-02-17:
The problem is that the BelHeli program does not see the regulators and does not detect them!
tried different versions of BelHeli programs .. both from the BLHeli Configurator computer, and from Google launched - any of these options after clicking read
 shows nothing!

BLHeliSuite opens, sees that there are BLHeli_32 regulators and that's it! - no more imformation, there is also no definition of the sequence of engines ...

BLHeli 32 - starts and shrinks ... without opening!
 Flight flyer
 Matek F405-STD
 regulators
 HAKRC BLHeli_32 Bit 35A 2-5S ESC

4712, 2021-02-17:
@Krilatii 
What version of BLHeliSuite32 do you use? It starts and shrinks, without opening?
What operation system?
Which iNav version is on the FC installed?

Krilatii, 2021-02-17:
Hello.
1) the version is downloaded from your site
https://github.com/bitdump/BLHeli/tree/master/BLHeli_32%20ARM
2) yes! it is this version that starts and contracts without opening.
3) Win 10
4) run  google chrome
I did the installation blheli 32 on another computer and it worked! but for some reason this is the case on my machine

4712, 2021-02-18:
@Krilatii BLHeliSuite32: probably you did not extract the zip file before starting the exe.

Krilatii, 2021-02-18:
And where is the zip file you are talking about? Where can I download it from?

4712, 2021-02-18:
@Krilatii you can find BLHeliSuite32 here:
https://drive.google.com/drive/folders/1Y1bUMnRRolmMD_lezL0FYd3aMBrNzCig
or here:
https://www.mediafire.com/folder/dx6kfaasyo24l/BLHeliSuite

BLHeliSuite32_32721.zip is the zip file to extract.
See [here](https://support.microsoft.com/en-us/windows/zip-and-unzip-files-f6dde0a7-0fec-8294-e1d3-703ed85e7ebc) on how to unzip the zip file.

Krilatii, 2021-02-18:
I thought that there was some kind of special file and installation method)) that's exactly what I did. thank

4712, 2021-02-18:
And does it work now?
If not, is there any dialog with error message shown?

Krilatii, 2021-02-18:
I did this the first time!) and it did not work, but minimized, and there is no dialogue about the error (

Krilatii, 2021-02-18:
Where to look for the problem? in the very win 10? Or is it a software error?

4712, 2021-02-18:
I cannot reproduce it here on Win10.
Can you provide a screen shot of the app in minimized state?
Do you have multiple displays?

Krilatii, 2021-02-18:
I have one screen. this process is not displayed anywhere after .
after clicking on the start file, the launch wheel scrolls .. and nothing happens

4712, 2021-02-18:
Could you please extract the zip file attached and copy the
BLHeliSuite32.ini into the BLHeliSuite32\Settings\ folder and start the app again?

[BLHeliSuite32ini.zip](https://github.com/bitdump/BLHeli/files/6003338/BLHeliSuite32ini.zip)

Krilatii, 2021-02-18:
I have done this operation. she did not help (still nothing ... thanks

4712, 2021-02-18:
Please exchange the attached BLHeliSuite32.exe (it excludes scanning for HID devices) and retry.
[BLHeliSuite32Beta.zip](https://github.com/bitdump/BLHeli/files/6003554/BLHeliSuite32Beta.zip)

4712, 2021-02-18:
I can't imagine, how disabling win updates could affect BLHeliSuite32.