bitdump/BLHeli #507: Brake not adjustable if PWM goes below min throttle setting

aaromalila, 2021-01-17:
In F5J-class a motor time limiter is used between RX and ESC. It forces throttle PWM to ~800µs (0.8 ms) after certain period of time.
This "shut down" -setting causes ESC (Holybro Tekko Metal) to apply full brake force **which damages the transmission gears**.

Brake setting % has no effect on this phenomena. Brake setting works fine if lowest PWM  is eg 1.0 ms, but the 0.8 ms causes this problem. 

I suggest that there would be such an option, that **brake force would not be dependent on PWM value applied**; the brake value should be always the same, even PWM was less than minimum throttle. This would make the BlHeli32 yet better in our competition class.

sskaug, 2021-01-17:
Ok, so this is probably because at around 800us, BLHeli_32 considers the pulse illegal and flags signal as lost.
I acn make you a test code where this limit is lowered. What is the exact name of the firmware that your ESC uses?

aaromalila, 2021-01-17:
Hi,
Thank you for a fast reply.
The ESC is a Holybro Tekko F3 Metal, firmware seems to be 32.7
[image: image.png]

I made 2 short videos about the problem. In the first one the brake behaves
ok as I manually control the PWM to 1.0 ms:
https://youtu.be/BSSfQ0ArTUg

We like to use quite smooth brake setting in order to save the gears and
motor; max power in some setups is almost 1 kW and motor can be spinning
70000 rpm. This is reducted to ~10000 with a transmission which really does
not like very quick stops :)

In second video I let the F5J-logger do it's job; it shuts down the motor
when 30 seconds is gone: https://youtu.be/RUmma4f5mQo

The logger inputs 800 microseconds pulse (0.8 ms) as the rules state that
"motor must be reliably shut down".  We can not affect on that.

Best regards,
Aaro Malila, Finland

sskaug, 2021-01-18:
I now put a testcode here that accepts down to 750us pulses as valid signal:
https://github.com/bitdump/BLHeli/tree/master/BLHeli_32%20ARM/Pwm%20down%20to%20750us%20testcode

aaromalila, 2021-01-18:
Hi,
Thank you. I will test in the evening. Is there some guide to install the
se-version? Can I use BlHeliSuite?

Br
Aaro

aaromalila, 2021-01-18:
Hi
Sorry to bother you with probably quite a trivial topic, but I have not
managed to proceed with the flashing.

BlHeliSuite detects the ESC (and has all kinds of flashing options,
including adding the file), but it tells the user to use BlHeliSuite32
instead, from where I'm not able to find a function to select a hex-file or
anything like that.

Also, blHeliSuite asks for .eep -file, not hex.

Could you please give some quick hints how to proceed. I'm of course
browsing all kinds of manuals and tutorials but without success...

Br.
Aaro

aaromalila, 2021-01-18:
... and immediately after my previous post I found some guidance and
managed to proceed with the hex-file. Following error was given:
[image: image.png]

Maybe I did not elaborate, that the Tekko is the single ESC version "B" ?
Anyway, I'm not able to proceed from this point. Sorry :)

Aaro

sskaug, 2021-01-18:
The image above does not show.
Again - what is the exact name of the firmware that your ESC uses?

aaromalila, 2021-01-18:
Hi,
Ok. The missing image shows the error information displayed when I loaded
the HEX file in flash-dialog:
"Current target ESC#1: TEKKO32_F3_Metal_B".

BLHeliSuite_32 tells as ESC information that current firmware version is
"TEKKO32_F3_Metal_B - Rev. 32.7 - Multi"

Br.
Aaro

sskaug, 2021-01-18:
Ok, new testcode now posted.

aaromalila, 2021-01-18:
Thank you again, I will test tomorrow evening.
Br.
Aaro

aaromalila, 2021-01-19:
Hi again,
I tested it, no success; There seems to be some inconsistency between what
is detected and what is expected.

This time I took screen captures from each phase and added those to a
Dropbox-transfer; you can download the picture here:
https://www.dropbox.com/t/YyYLhpQe6nLHyvGp
(just to avoid typos when copying the error information).

I hope that helps :)

Aaro

sskaug, 2021-01-19:
Are you sure you select the right file?
For me, the github file flashes fine:
[image]

aaromalila, 2021-01-19:
Pretty sure - I deleted the old local file before downloading the new.
Maybe I still try again.
The error was about the ESC not being of the version to which the FW was
ment, if I understood it correctly..

I'll go and test. I download the file from this location:
https://github.com/bitdump/BLHeli/tree/master/BLHeli_32%20ARM/Pwm%20down%20to%20750us%20testcode

aaromalila, 2021-01-19:
Hi again
Still same error after selecting the file:

ESC#1: Flash "TEKKO32_F3_Metal" failed in 0.00 s
Found different ESC Layouts!
Current Target ESC#1: TEKKO32_F3_Metal_B
Hex File is for: TEKKO32_F3_Metal

I tried also wit another (identical) Tekko-ESC, same behaviour.

Br,
Aaro

sim-, 2021-01-19:
Hello! :) Just a note, I used 768µs as the minimum pulse length because I saw at least one thing outputting 800µs for idle. It has been a few years, though, and I don't remember what. It may have just been a servo tester.

aaromalila, 2021-01-20:
Hi
Ok. In my case this is not possible as the f5f-logger between rx and esc
requires 1.2 ms pulse to detect "motor start" event. If min pw is set much
lower, there will be a big domain where the logger does not work correctly.
Aaro

sskaug, 2021-01-20:
Hi Simon! Thank you for the info, good to hear from you again.

sskaug, 2021-01-20:
https://raw.githubusercontent.com/bitdump/BLHeli/master/BLHeli_32%20ARM/Pwm%20down%20to%20750us%20testcode/TEKKO32_F3_Metal_B_Multi_32_74.Hex

aaromalila, 2021-01-20:
Hi again,
Now I managed to flash the ESC with the test version. The original way to
flash ended up with same errors as earlier, but after some inspection I
tried a workaround and it worked. Here is a screen capture of the success :
https://www.dropbox.com/t/jnBxWCYCC1Bez7ee .

Anyway, the result was not yet ok. Despite of whatever brake setting
(starting from 0%) the behaviour is always a "full stop" when PWM goes to 1
ms or whatever is set as low limit. Brake setting has no effect on this, it
seems that the brake function is actually not working at all.

Please tell if you want me to do some specific tests. I have a protocol
analyser / oscilloscope and I already verified from signal wire that PWM is
what I wanted it to be, 1000 - 2000 µs, and there are no anomalies.

Br,
Aaro

sskaug, 2021-01-20:
Can you share the ESC settings that you use (e.g. screenshot of BLHeliSuite32?
Do you run nondamped?

aaromalila, 2021-01-20:
Yes, I will send you a picture tomorrow.
I remember the settings that differ from default:
No damping, auto timing, slowest possible acceleration, min thr 1.2ms, max
thr 2.0ms.

I tested today min thr 1.0 and checked that my radio also provides that.
Originally 5..15% brake, but now it has no effect at all.

Pulse train repetition interval is 17ms.

I can test also with another Esc-unit tomorrow in case the one used until
now is somehow bricked. I have 3 identical units.

Basically, as more or less pseudo code, could the issue be solved with eg.
a quite early filtering block like

If (pwm_in < 1e-3)
   {
   pwm_in = 1e-3;
   }

Br
Aaro

aaromalila, 2021-01-21:
Hi,
Here are rest of the settings: https://www.dropbox.com/t/0Ag1r29HZcu4LMhH
I tested with another ESC, same behaviour.

Br.
Aaro

sskaug, 2021-01-21:
In these settings you have nondamped mode off. That will give you quite hard active braking.

aaromalila, 2021-01-21:
Hi,
Ah. Ok  The logic has to be read and thought carefully; disabling
non-damped mode activates the damped mode :)

Sorry if I wasted your time with the non-damped-setting, it is possible
that it has been wrong during these last tests.

Now I tested with test-sw and non-damped mode on. ESC behaves ok, no sudden
stop on 1.0 ms or 0.8 ms, but brake adjustment feels somehow binary. Till
~80% brake there is no effect at all, and with 80% or more, eg 100%, there
is some.

Aaro

sskaug, 2021-01-22:
Yeah, it's a double negation - awkward, but with some history.

So the new test code is ok, but the Rev32.7 code misbehaves?

Makes sense that the braking effect is quite nonlinear.

aaromalila, 2021-01-22:
I will test and cross test with few different setups and units. Brake
behaved earlier OK and adjustment was linear, but it might be also possible
that the esc-unit used has somehow damaged - I noticed yesterday that the
system has some hick ups when applying more power. These did not occur
earlier.

Br
Aaro

aaromalila, 2021-01-22:
Hi
Now I tested the 32.74 with another setup and I guess it works quite ok.
There is no sudden stop, and in the end, brake works well enough for our
purpose.  This is currently based on ground testing only as flying season
will start no earlier than in March...

For some reason the brake behaviour vs. adjustment value has changed, with
fw32.7 (and legal input pulse) the adjustment was rather linear, but now
with 32.74 it brakes only if set to 100%, after a ~0.5...1 seconds free
spinning. But anyway, this is fine as there is the braking effect in the
end, and it is strong enough to stop the prop and let the blades fold back.

(without brake the blades keep spinning due to airflow, and causes
aerodynamic brake effect which ruins the glide...)

Can I share the test version with a few friends, or should they wait for
possible "official" release?
And is it ok to discuss this topic / testing on some forums related to our
FAI F5J class?

Thank you very much for your efforts.

Best regards
Aaro

sskaug, 2021-01-23:
Thanks for the feedback, I will then implement the minimum limit change from 800us to 750us on future codes.

Sure, please go ahead and distribute code and discuss!

aaromalila, 2021-01-23:
Ok, thank you.
Aaro