bitdump/BLHeli #513: Question: Where are the definitive specifications for dShot and Bidirectional dShot?

jimkazmer, 2021-01-27:
I've implemented both protocols using a Teensy 4.0 microcontroller.  When I reviewed my notes, I found some discrepancies.  I started searching the web looking for a spec for each (Dshot and bidirectional Dshot), and am not finding anything complete.  

You'd thing there would be a spec document available for each one.  Will Dshot be supported long term? Will bidirectional Dshot become the only supported Dshot?

Thanks,

mathiasvr, 2021-01-27:
I also haven't found a complete specification but got most info about bidirectional dshot from  https://github.com/betaflight/betaflight/pull/8554#issuecomment-512507625 with regards to telemetry.

jimkazmer, 2021-01-27:
Thank you... I had seen that post in the past, but could not find it again.  That is the most detailed info that you can find anywhere.  Do you happen to know what type of CRC is used for the non-bidirectional dShot?

This is the best I have seen for regular dShot: https://blck.mn/2016/11/dshot-the-new-kid-on-the-block/

jimkazmer, 2021-01-27:
This is what I found for the checksums:
//calculate checksum
csum1 = (value ^ (value >> 4) ^ (value >> 8)) & 0x0F;  // Non-Inverted Checksum
csum2 = (~(value ^ (value >> 4) ^ (value >> 8))) & 0x0F;  // Inverted Checksum

mathiasvr, 2021-01-27:
Yes the checksum is the "nibbles" xor'd together. For bidirectional dshot the checksum is inverted.
So in your example `value` would be the 12-bit dshot value (11-bit throttle and 1-bit telemetry request)

closed 2021-01-30 by jimkazmer

elmagnificogi, 2023-04-19:
I have wriete a guide about Bidirectional DSHOT and DSHOT, if someone need migrate code could refer it.
It's chinese but you could translate

> https://elmagnifico.tech/2023/04/07/bi-directional-DSHOT/