bitdump/BLHeli #517: Motor keeps spinning if PWM signal is unconnected

mice5, 2021-02-02:
Hi!

I have observed that If i spin the motor with, let's say 1200us PWM (I'm using 1000-2000us 400Hz PWM signal), and disconnect the signal wire, the motor keeps spinning with additional 1-2Hz stuttering (like resetting the lost signal timeout) until power source is disconnected.

Moreover, If i disconnect 1000us brake signal, motor can occasionally spin-up to some random speed until i reconnect the signal wire or power is lost.

Using BLHeli32 32.7.

Regards,
Mateusz

sskaug, 2021-02-02:
When the signal wire is disconnected, generally the internal pullup of the MCU is used to tie the input wire. This pullup however is not strong, so in some conditions where there is external noise (motor running, touching the wire etc), a valid signal can be seen. Therefore always remove props when working with the system

mice5, 2021-02-04:
Could be some noise isse... but:
- Retested with unpowering the signal source / signal wires connected.
- Retested with manually shorting the esc signal pad to ground
- Retested with manually shorting the input signal pin directly on the MCU
- Retested with disabling the signal sent by the source.
Every test results in unwanted motor spinning.

sskaug, 2021-02-07:
I can not reproduce this with my ESC here.
Which name does BLHeliSuite32 give for the code on your ESC?

mice5, 2021-02-08:
Exuav_FishDrone-Rev.32.7 - Multi

sskaug, 2021-02-08:
I am still not able to reproduce the behavior you see.
But - I did find a weakness in the code, in handling signal loss.
So updated 32.7.4 code is now posted.

mice5, 2021-02-09:
I've flashed 32.74 testcode and it fixed the problem.

Then, for additional confirmation I flashed it back to 32.7 - problem exists. And again, to 32.74 - problem fixed.

Even if it does something weird, like spinning to the max, it switches completely off after the timeout. Before the fix, it would always spin with 2Hz stuttering.

> I am still not able to reproduce the behavior you see.

Maybe it's a matter of some setting, like auto telem or so:
[image]

sskaug, 2021-02-09:
Great to hear - thank you for reporting and testing.
My guess is that it's brake on stop or maybe stall protection that causes the different behavior from my default settings.