bitdump/BLHeli #519: ESC telemetry voltage sensor issue

ghostkcleung, 2021-02-24:
The blheli_32 version 32.7 is flashed into my Zeez 4in1 60A ESC. But there is some problems with the reading of voltage and current. It's OK when read with ADC. But when read with esc sensor. The current is about 6-7A while there is no any load. Is it possible that is the issue of the ESC firmware ?

[image]
[image]
[image]

sskaug, 2021-02-24:
We have never heard of any such generic issue, so you should probably contact your vendor about this.

ghostkcleung, 2021-02-24:
Is it possible that the problem similar to the following issue?After I disable the bi-direction in the betaflight configuratior. The current reading is below 5A although it is still has error.

https://github.com/bitdump/BLHeli/issues/336

Jahnkeanater, 2021-03-24:
> 
> 
> Is it possible that the problem similar to the following issue?After I disable the bi-direction in the betaflight configuratior. The current reading is below 5A although it is still has error.
> 
> #336

Possibly I'm having a pretty similar issue. My 4S ESC is reporting 17-18v from the esc and the current is +/- 10 amps. I'll try turning off bidir dshot and see what that does.

Jahnkeanater, 2021-03-25:
I just updated to the latest development build of betaflight 4.3.0 and the issue was fixed. This issue must be in betaflight 4.2.8.