bitdump/BLHeli #520: BLHeli crashes PC when trying to connect

rac4433, 2021-03-01:
I've downloaded the BLHeli32 software suite from 2 different locations and the results are the same on two different PC's
One PC is a Win 7 laptop the other PC is a Win 10 desktop .
This BLHeli32 software will not run on either of them .
It crashes PC when I try to connect to ardupilot flight controller .
I've tried using both Matek and Holybro flight controllers , the results are the same .

I have to CTRL+ALT+DEL to get back control of PC when it crashes . I'm able to connect either PC to any FC using 
Mission planner so I don't have any virtual .com port driver problems

4712, 2021-03-01:
You will notice that the PC becomes unresponsive, which will stop as soon, as you disconnect the BC from USB.
It is caused by trying to clear serial buffer, and ardupilot is sending data to the port without breaks.  
Which version of BLHeliSuite32 are you using?
Which FC is it exactly?
Please post the ardupilot param file here.

4712, 2021-03-01:
@rac4433 Please download the BLHeliSuite32 rev 32.800 from one of the links [here](https://github.com/bitdump/BLHeli/tree/master/BLHeli_32%20ARM#blheli_32-downloads). Attached is a BLHeliSuite32,exe beta version, with better (no hang) behavior. Please exchange the exe and try.
[BLHeliSuite32.zip](https://github.com/bitdump/BLHeli/files/6064596/BLHeliSuite32.zip)
But, I guess, you have something wrong in your Ardupilot parameters and it will not connect either.
Please check https://ardupilot.org/copter/docs/common-blheli32-passthru.html

rac4433, 2021-03-14:
Hello    Your file won't download . I get this google drive screen which I know nothing about and click on your file and nothing happens .I get a small message in lower left corner saying "scanning for virus" but the file never gets to my computer . I search for BLHelisuite32_32801.zip and its not here  .
R Champagne

4712, 2021-03-14:
@rac4433 no problem here, downloading the file... Which browser do you use?

haslinghuis, 2021-03-14:
@rac4433 On `firefox` you have to click in the middle or on the name to download - clicking at the icon on the top left does the virus scan. It's not intuitive. With `chrome` you can right click and choose Download in the menu.

[image]

rac4433, 2021-03-15:
Hello    I have a question about configuring the setup . Somebody else I was talking to mentioned problems using BLHeli32with MAVlink .Right now I send everything ( RC command and telemetry ) over a single MAVlink configured serial port (SERIAL 1 in this case)BUT . I also have SERIAL 0 ...(usb console) set up the same way . SHould I reconfigure SERIAL 0 to -1 (default) instead of 2 (MAVLink 2) ?

Thanks for the help
R Champagne

haslinghuis, 2021-03-15:
Please check https://ardupilot.org/copter/docs/common-blheli32-passthru.html

rac4433, 2021-03-16:
Hello    Ive been thru that article at least 20 times by now . This BLHeli32 simply does not work with the software/support documentationthat is available to me .    Bottom line is I want to get away from ardupilot completely ( at least where digital esc's are needed ) and use something like INAV .I get that INAV has fewer features but what good are they when the documentation is out of date and they dont work ?    Besides , you wouldn't have to listen to my dumb questions anymore !At this point I want to reflash the ardupilot FC to INAV .Problem is I cant do that because ardupilot wiki on how to do that is out of date and doesn't work !!!I can't remove ardupilot from EEPROM .Factory reset doesn't work in mission planner !!WHen you try to connect using terminal ( necessary to issue reset commands) you get endless streams of charactors rolling down the screen ! Some sort of MAVlink problem that I don't understand and is not explained anywhere .
If I could wipe this ardupilot off the FC I can move on . Ill probably end up dumping in trash , only $50 not the end of the world I guess .

4712, 2021-03-16:
@rac4433 here are the standard parameters for MatekF765Wing arduplane 4.0.9, but with BLHeli 4way-interface enabled.
[MatekF765Wing-arduplane-BLHeli_32-passthrough.zip](https://github.com/bitdump/BLHeli/files/6149674/MatekF765Wing-arduplane-BLHeli_32-passthrough.zip)
Maybe you try this parameters and merge yours into it step by step after compare.