bitdump/BLHeli #522: No current telemetry from Lumenier Razor 30A

MattxFR, 2021-03-10:
Hello,

Issue 1: no telemetry reading from current sensor, 0A value wathever I do, in BF, OSD, blackbox.
Issue 2: No option to limit current in Blheli_32, firmware 32.7 although I see this option in the user manual and some other videos from JB etc.

I do have ESC voltage telemetry and ESC temperature so I know my telemetry is set correctly in BF.

I would think the two issues above are related but not quite sure. Other than that the ESC is working fine, but I really want the current to see how much I can push staying safe from frying the ESC..

FYI: From the same brand/serie I had two faulty FC (Lumenier F7 mini) until i changed to another brand. I would like someone to confirm if the above is symptomatic of yet another bad quality hardware or if there is anything you would think of?

Kind regards, Matt

sskaug, 2021-03-10:
As you do not see the current limiting setting, this means that your ESC does not have individual current sensor on each motor. Not all ESCs have that.
Some 4in1 ESCs have a current sensor that can feed an analog signal to the FC - maybe your ESC has that?

MattxFR, 2021-03-10:
Hi THanks for your reply, i should have the analog and the telemetry for the current sensor but both are readin zero..
I found other people having current sensor dead on this brand so that might be it ..

Matthias Favrel

Envoyé le :mercredi 10 mars 2021 12:49
À : ***@***.***>
Cc : ***@***.***>; ***@***.***>