bitdump/BLHeli #523: Motors spin eratically  when connected to blheli32

airmigs, 2021-03-12:
While in Blheli32 suite when a lipo is plugged in the motors spin erratically with Bidirectional Dshot
when Bidirectional Dshot is disabled its fine but esc has bad input frames
tekko32_f3_c metal 4-1 65a esc, radix 2 flight controller

4712, 2021-03-12:
@airmigs Bidirectional Dshot may complicate the connection to the ESC bootloader. Please try to first click "Connect", then connect the power source to the ESC, wait all beeps, then click "Read Setup" or "Check".

To help investigate further improvements: 
What BLHeliSuite32 version do you use?
What BLHeli_32 version do you use on the ESC?
What Betaflight version do you use on the FC?
What Betaflight target name do you use on the FC?

airmigs, 2021-03-14:
using BLHeliSuite32.8.00
Tekko32_f3_C Rev. 32.7
Betaflight 4.2.5 from Brainfpv
RADIX2
I've tried the Radix2 on another kwad and same issue, double checked my wiring harness
I used another FC to calibrate the tekko32 esc
flew the kwad with radix2 tekko32 set up on stock betaflight setting wi bidshot on and well heres a pic of blackbox log
[image]

4712, 2021-03-14:
@airmigs does this mean, it's only an issue with RADIX2 and a custom Betaflight version from Brainfpv?

airmigs, 2021-03-15:
I agree with you 
So I flew again today with bidirectional dshot disabled and the kwad flew ok on stock betaflight settings 
With the radix 2 fc pid loop is max at 3.2k kinda weird for h7 fc
Ill continue testing, already tried to warranty the fc maybe mine is faulty when bidirectional dshot is on or I don't have the right settings