bitdump/BLHeli #524: BLHeli32 app revision 1.0.22 AND PC app will not connect to arduplane on MATEK F765 (4.0.9)

rac4433, 2021-03-12:
The ongoing saga of not being able to connect to BLHeli32 esc continues ......
AFter much effort I cannot connect to any BLHeli32 esc using ANY flight controller running ardupilot
Nothing will connect , not the android app , not the PC app . nothing works 
The same computers and phones can connect to flight controller or mission planner or Q ground control fine .
There is nothing wrong with usb drivers .
These BLHeli32 esc's are completely useless !!!

4712, 2021-03-12:
@rac4433 we understand that you are in a frustrating situation, but without answering questions it is hardly possible to help. 
How about continuing at #520 first.
Please try and report your experience with [BLHeliSuite32.zip](https://github.com/bitdump/BLHeli/files/6064596/BLHeliSuite32.zip) and please post the ardupilot param file for MATEK F765 here. 
Most likely your parameters are not setup to work with BLHeli_32 Pass-Through.

rac4433, 2021-03-12:
Hello
The reason I abandoned the PC version of BLHeli32 is because it has the same problem and nothing works .Understand the esc/motor does work fine . Ive tried both Matek FC's and Kakute F7 AIO V1.5 , same problem .The only common denominator is arduplane (4.0.9) .As mentioned I have this all put together on bench . The latest setup is MATEK F765 with dual cameras and DVR record on/offcapability . Everything runs fine . I can power up the motor to full throttle , pulls a little over 20 amps with the 1806-2400Kv motor and 5 inch prop .Cameras , switching , lost model beeper , OLED display , Vtx with smart audio ( controlled direct from TBS nano RX ) , it all works . My first time using dual cameras . The relay configurations went smooth .Also , mission planner is disconnected before any attempt to connect to esc .
With the ardroid app I get error at connection attempt MSP_API_VERSION failed . however I can connect to FC fine using Qground control app . When I plug usb into FC I get a message asking if I want to connect to MATEK F765 ? So my phone can "see" the flight controller . I found one post from a user with same problem . His issue was a setting problem . I checked my settings and they are correct .

MATEK F765 wing setup

I have PWM outputs 1 and 2 ( group 1) configured as Dshot 150 . Im only using #1 for throttle but wiki says I have to configure every output in a group as Dshot ( in this case #1 and #2) . 
SERVO_BLH_PORT=0SERVO_BLH_MASK=3 ( CH1 AND 2 , esc connected to CH1) Ive tried it set at 1 , no differenceSERVO_AUTO_TRIM=0SERVO_BLH_AUTO=0SERVO_BLH_OTYPE=4 (DSHOT 150)
Im at a complete loss here .There really isnt that many setting involved from what I see ?I just can't connect to change settings .I tried flying one wing ( good thing it was a small cheap one ) without any esc tuning and it lost power after takeoff and now sits at bottom of Quabog river !
If I sound frustrated I am . Got home earlier after a good hike up a mountain to fly my DJI phantom 4 pro v2 . Got to the top ( where Ive flown from before ) and phantom would not take off , some new geofencing bullshit !
Thanks for the prompt reply 

R Champagne

4712, 2021-03-12:
@rac4433 could you please provide the .param file here (saved with Mission planner), so I can load it to my F765 here?

4712, 2021-03-13:
@rac4433 Sorry can see no file here, I guess you need to post it as zip.

rac4433, 2021-03-13:
Hello
Attached is .param zip folder for Matek F765 setup .I'll include my crossfire TX/RX config here 

1) MAVTX2) MAVRX3) CH34) Smartaudio5) CH56) RSSI7) CH78) CH89) CH910) CH1011) CH1112) CH12
RC by MAVlink  ON12 CH MODE

TARANIS X9D+2019 setup below (ELEVON WING CONFIGURATION)

CH1 ROLLCH2 ELEVATORCH3 THROTTLECH4CH5 LOST MODEL BEEPERCH6 RSSICH7 GPIO CAM TRIGGER (RELAY)CH8 FLIGHT MODECH9 Vsw PAD POWER( C2 CAM PWR) RELAYCH10 C1/C2 CAM SELECT
Also there is a strange bug in mission planner where if you open the optional hardware screen and pickcameras and gimbals the CAM_TRIGG_TYPE setting always changes to servo on its own . Weird , this caused great frustration a month ago but Im aware of it now . Once again say you have CAM_TRIGG_TYPE set to RELAY and you open the cameras / gimbals page , when you leave mission planner you'll find CAM_TRIGG_TYPE has been changed to SERVO on its own ! You have to make sure your CAM_TRIGG_TYPE setting is where you want it .....STAY AWAY FROM CAMERAS AND GIMBALS PAGE ....then you can disconnect and unplug the FC . 

Hope this helps
Thanks againR Champagne

4712, 2021-03-13:
@rac4433 again no file is attached, you may need to paste it with the browser at Github webpage.