bitdump/BLHeli #526: odd behaviour of BLHeli_32 max acceleration

cbf123, 2021-03-30:
I found an interesting Youtube video at https://www.youtube.com/watch?v=GNFcWww5RPI discussing how the "max acceleration" behaviour of BLHeli_32 doesn't actually behave the way it's documented to behave.  In particular the acceleration is not linear in either the RPM domain or the power domain, and the damped deceleration is not affected when logically it probably should be since it's essentially a negative acceleration.

Is he right, or is there something else going on?

Also, the docs say that max acceleration limits the _power_ applied to the motor...is that accurate or is it actually supposed to limit the rate of increase of the motor RPM?  (Since the power increases nonlinearly with increasing RPM, and not all ESCs have current sensors which would be needed to reliably limit power.)

sskaug, 2021-04-01:
I find this video to be inaccurate and misleading in some ways.
BLHeli_32 max acceleration regulates the ramp of the power applied to the motor.
Which does not guarantee a linear ramp in RPM, nor an immediate settling to any given RPM.

The purpose of this programmable setting, is to be able to limit instantaneous power increase steps applied to the motor, that could otherwise cause corruption of the BEMF signal and loss of sync. So you would limit acceleration only to a level that is not noticeable unless you apply a very abrupt power step. And limiting deceleration or braking has never been the purpose.

cbf123, 2021-04-01:
Thanks for the quick reply.  The stated goal of the max acceleration option is to "prevent desync on some motors under high acceleration".  Is it possible to get desync under high regenerative braking force as well?  If so, it seems like we would want to apply the same setting to limit regenerative braking.

How are you calculating the power applied to the motor when not all ESCs have current measurement?

If the idea is to linearly ramp the power applied to the motor, then I'd expect the RPM to exhibit a behaviour where it increases quickly at first, but then the rate of increase slows down and eventually stops increasing when the ESC reaches wide-open-throttle.  (Because as the RPM increases, adding a given amount of power will result in a smaller increase in RPM.)

Instead, the graph in the video shows that as time progresses the RPM increases *more* for each increment of time.  This means that power is increasing much faster than linearly with time.