bitdump/BLHeli #527: Impossible to write or flash ESCs

flixiflux, 2021-04-13:
Hello,

Sorry but after searching for weeks and trying lots of things I’m really lost and would really like to be able to flash my ESCs. I want to put it in 48hz but the most important, be able to use turtle mode (motors don’t reverse when turtle mode).

I bought a US65 (crazybeeF3FR) to somebody but it’s totally impossible to flash ESC, in the configurator you can’t save anything, even if I just want to write a simple setting:/ I doesn’t work in jesc configurator, jazzmaverick, blheli configurator, blheli suite.

Here is the log error when trying to flash. If someone could give me advises it would be great as I really worked on it for hours now:(

Thanks for reading me :)
```

ESC 4: flashing <strong>started</strong>
sending cmd_DeviceInitFlash 0 3
received cmd_DeviceInitFlash ACK_OK 0 177,232,100,1
sending cmd_DeviceRead 1a00 112
received cmd_DeviceRead ACK_OK 1a00 16,5,33,255,255,255,255,255,255,11,255,1,255,85,170,1,255,255,255,255,255,3,255,255,255,37,208,40,80,4,255,2,255,122,255,7,1,255,255,0,0,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,35,79,95,76,95,48,53,35,32,32,32,32,32,32,32,32,35,66,76,72,69,76,73,36,69,70,77,56,66,49,48,35,32,32,32,32,32,32,32,32,32,32,32,32,32,32,32,32
sending cmd_DevicePageErase 0 13
received cmd_DevicePageErase ACK_D_GENERAL_ERROR 1a00 13
163552.1800001152 resending cmd_DevicePageErase 0 13
163558.29500011168 resending cmd_InterfaceTestAlive 0 0
163558.60499991104 resending cmd_InterfaceTestAlive 0 0
163558.77000000328 resending cmd_InterfaceTestAlive 0 0
received cmd_DevicePageErase ACK_D_GENERAL_ERROR 1a00 13
163569.3550000433 resending cmd_DevicePageErase 0 13
received cmd_DevicePageErase ACK_D_GENERAL_ERROR 1a00 13
163583.0099999439 resending cmd_DevicePageErase 0 13
received cmd_DevicePageErase ACK_D_GENERAL_ERROR 1a00 13
163605.34500004724 resending cmd_DevicePageErase 0 13
received cmd_DevicePageErase ACK_D_GENERAL_ERROR 1a00 13
ESC 4: flashing <span style="color: red">failed</span> - <strong>Error: cmd_DevicePageErase ACK_D_GENERAL_ERROR
    at recvCallback (chrome-extension://maigancljkhnkhpejlfmgjjmjafafdhp/js/_4way_if.js:210:37)
    at chrome-extension://maigancljkhnkhpejlfmgjjmjafafdhp/js/_4way_if.js:316:29
    at Array.forEach (<anonymous>)
    at Object.onread (chrome-extension://maigancljkhnkhpejlfmgjjmjafafdhp/js/_4way_if.js:298:18)
    at read_serial (chrome-extension://maigancljkhnkhpejlfmgjjmjafafdhp/js/serial_backend.js:234:15)</strong>
```

mathiasvr, 2021-04-13:
Looks like the ESC might be locked. If that's the case you have to erase it using the C2 interface.

flixiflux, 2021-04-13:
Thank you very much for your answer. 
Yes it seems the ESC are locked, that would be so great if I could unlock them, but sorry for my n00bness, what is the C2 interface ?

flixiflux, 2021-04-13:
Ok, google is my friend, I see what it is, I'm gonna try to learn that and see if I can sort it. If you have a tutorial you know is well done don't hesitate to telle me. Thanks again, very kind from you to answer :)

mathiasvr, 2021-04-13:
It's not straightforward and requires a debug adapter or you can make c2 programmer using an arduino, etc.
You will need to access the c2d and c2ck pins of the mcu (efm8), if you are lucky they have been traced out on your board.

flixiflux, 2021-04-13:
Ok, it seems like it's gonna be a big adventure ^^ But I like to learn, and I have an arduino, I'm only frighten that I have to solder on a very little FC, I hope it's on an easy place. I don't understand how the guy who sold me this drone managed to lock the ESCs, sure that's the reason he sold it :( Do you think this could be risky ? Because the drone flyies great, but no turtle mode and no possibility to put it in 48hz, on the other way, if I break it it will not fly at all ! To sum up, do you think this is something to try for somebody not really experimented in those kind of things ?

flixiflux, 2021-04-13:
If think I'm gonna let my drone like this, it seems crazybeeF3 is not documented to know where c2d and c2ck are on the board :/ Anyway thank you very much for your help, I'm happy to know, even if I can't do anything to solve it, perhaps this post will help other persons with other board where c2d is findable.

mathiasvr, 2021-04-13:
Yeah, if you don't want to risk damaging it I don't think it's worth it. I might be able to help you figure out the pinout if you want to try anyway though? 😅 But in that case, you will probably want to see if you can make the c2 programmer first.
Is this just a problem with 1 esc or all of them btw? And which version of blheli_s is it running?

flixiflux, 2021-04-13:
All 4 ESC are locked, it runs 16.5 for the moment, I've read that's the reason why turtle mode doesn't work. The very strange thing is that I have other US65 and all of them have turtle mode out of the box, and I'm not sure but I think they were also 16.5 (can't say because I flashed them). 
My main problem is clearly turtle mode, so perhaps it's possible to find another way to solve that, I think it's more reasonnable not to try the operation, I know myself and I'm sure I'm gonna kill the FC... Thanks a lot anyway :)

mathiasvr, 2021-04-13:
Yes, looks like 16.5 doesn't have support for turtle mode ☹️ 
And yeah if it's all 4 ESCs it's going to be a lot of work/risk.

flixiflux, 2021-04-13:
Ok, thank you very much Mathias, I close the issue :) I guess the guy who solded it to me did crap with the ESC as US65 works with turtle out of the box...

closed 2021-04-13 by flixiflux