bitdump/BLHeli #528: Blheli_32 motor test mode fail

Roxxtar1, 2021-04-21:
Hello,

Whenever I try to test my motors in blheli 32 and as soon as I click on the warning "I understand etc..." I get the below error.

Initialization of Motor Test Mode failed! 

Please check current unknown Rev. 36.77. 62 support for this application function! 

I am on the latest blheli 32.8 version.

Appreciate your support.

4712, 2021-04-21:
What FC with what firmware?

Roxxtar1, 2021-04-21:
Rush blade f7 with Rush esc 50a and flightone Falco x.

4712, 2021-04-21:
Motor test function is basically compatible with Betaflight.
Flightone seems to have implemented a subset of Betaflight msp protocol, but may be not the parts needed for motor test.

Roxxtar1, 2021-04-21:
OK thanks makes sense I guess

Jai-GAY, 2022-08-29:
i cannot get [HobbyWing XRotor Micro 60A 4in1 BLHeli-32 DShot1200](https://www.hobbywing.com/goods.php?id=653) to spin the motor in [the Motor test](https://youtu.be/np7xXY_e5sA?t=755).

Eep_FW_Main_Revision=32
Eep_FW_Sub_Revision=70
Eep_Layout_Revision=44

I also experienced weird behaviour where after disconnecting the BLHeliSuite32_32.9.0.4 software, I could not arm my quadcopter or use the Mission planner motor test.

Does anyone have good experience to share?

4712, 2022-08-29:
@Jai-GAY do you see any error messages? Perhaps it would help to supply a log file here (options ... enable log, save to file as log.tlg and post as zip here).
It depends on the interface part in AP, how after disconnecting restoring the normal port functions is done or not. Mostly it's recommended to reboot the FC after using BLHeliSuite32 for connecting the ESCs. Using Motortest alone should not require reboot.

Jai-GAY, 2022-08-30:
When I moved the slider, the physical motors did not spin.
I try to play with the Alt Speed Test, but nothing happens also.
I am using [Arducopter 4.2.2](https://discuss.ardupilot.org/t/copter-4-2-2-released/88387).

> Using Motortest alone should not require reboot.

Not happening, if I just used the BLHeli software motor test, then disconnect, then switch to the MP motor test, the Motor did not spin. But after a hardware reboot, goes straight to MP and does a motor test, it is spinning.

[image]

[Log_motor_speed_test.zip](https://github.com/bitdump/BLHeli/files/9448754/Log_motor_speed_test.zip)

4712, 2022-08-31:
This seems related to the way AP handles the msp calls.
Have you looked into the code of the [4way-interface implementation](https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_BLHeli/AP_BLHeli.cpp) in AP?