bitdump/BLHeli #529: Feature request - BLHeli_32 Suite usb connection seeeduino XIAO

hgo-be, 2021-05-01:
I tried to connect blheli_32 suite android to iflight Esc using Usb_toTtl adapter tied tx and rx together using a diode for one wire communication.
Usb connection is OK but error message pops up "no esc found" 
On the scoop the two way communication seems OK. 

I like to program a seeeduino XiAO board as an interface board.
      Usb connection Unknown device 10374/32815 ( seeeduino XIAO )

Request : enable usb connection with seeeduino XIAO (  vid/pid 10374/32815 )

sskaug, 2021-05-04:
Sorry, at the moment we have no plans for adding seeeduino XIAO as an interface board

hgo-be, 2021-05-05:
Ok. I could not found how I can configure a blheli32 with my android

4712, 2021-05-06:
You can simply use an Arduino Nano flashed by BLHeliSuite32 for BLHeli bootloader usage.

[image]