bitdump/BLHeli #530: Motor Max at 20% Speed using BL HELI ESC Versus 100% Speed using other ESC

mezerwi, 2021-05-03:
guys, need help here.
Recently bought thruster motor (3s brushless for my usv project)
I connect it with Turnigy Plush ESC - work flawless (full speed no issue)

So i bought flycolor blheli32 esc to make use of it bi-directional motor spin. 

it get the motor spinning, but at full throttle the motor only spinning at 20% speed
same issue happen on holybro tekko32 (same blheli 32). 

is there anything i need to check? why does it not spin full speed at full throttle?

I test using servo signal generator, toolkitrc servo signal generator and also pixhawk 4 which all yield same result.

sskaug, 2021-05-03:
This is most likely related to a low kv motor on low voltage. Try increasing rampup power, or alternatively turn low rpm power protection off.

mezerwi, 2021-05-03:
Thank you soo much!

Turn low rpm power protection off - solved the issue.

closed 2021-05-03 by mezerwi