bitdump/BLHeli #531: Feature request - S.Port telemetry name prefix/suffix

amordue, 2021-05-03:
My setup:
- Using a transmitter with OpenTX 2.3
- FRSky receiver
- Multiple blheli_32 escs (e.g. a 4in1 or multiple discrete single ESCs) connected directly to sbus for signal input and s.port for telemetry.

The s.port telemetry feature of blheli_32 is great when you have a single ESC connected directly to the receiver, you can uniquely identify the different sensors reported to OpenTX. 

OpenTX only displays the name, and not the "Instance" of a sensor. So you end up with for example 4 `EscA` telemetry sensors when using a 4in1 ESC, and no way to identify which ESC the sensor refers to, on the TX. 

Would it be possible to add an option to add a suffix to the names to make them uniquely identifiable? e.g. add the s.port physical port id value as a suffix, for sensor name values like `EscA1`, `EscV1`, `EscA3`, `EscV3`, etc.

Thanks very much!

haslinghuis, 2021-05-03:
Seems OpenTX is already able to do just that:

http://rcdiy.ca/aa-rename-a-sensor/

amordue, 2021-05-09:
> Seems OpenTX is already able to do just that:
> 
> http://rcdiy.ca/aa-rename-a-sensor/

That is true however it would be nice to not have to do that, for example, with a 4in1 ESC you'd have to rename 20 values. Doing this in the companion app is _okay_ but trying to do it on the TX itself is very very time consuming. Also if you need to rediscover sensors, e.g. you switch rx, then you have to do that all over again. Having the ESC automatically suffix the s.port id would be amazing.

haslinghuis, 2021-05-09:
@sskaug any room for implementation?

sskaug, 2021-05-10:
Actually none at all on the ESC side. The ESC ony detects a given physical ID and application ID and responds with 32bits of data. It has no way of naming any sensors, this has to be done in SW layers above.