bitdump/BLHeli #533: Question: Is it possible to disable Stall Protection?

AbhishekJoshi, 2021-05-17:
I am working on a use case, where I do not want stall protection enabled.

My setup will not burn. 

I can compile the code myself, but I haven't worked with assembly in years. If anyone can provide guidance on what changes need to be made, that would be appreciated.

I'm working with BLHeliS 30A ESCs.

mathiasvr, 2021-05-17:
Do you want to make the ESC continue attempting to start the motors when stalling?

There is also a stall boost feature that will apply more power on startup according to the number of consecutive stalls, do you want this disabled, or what is your purpose?

AbhishekJoshi, 2021-05-17:
> Do you want to make the ESC continue attempting to start the motors when stalling?

> There is also a stall boost feature that will apply more power on startup

I saw startup power, which I have max'd already, as well as temperature protection.

>  do you want this disabled, or what is your purpose?

I have a pair of simonk, and the behavior on that ESC is what I want to replicate on this BLHeliS ESC. 

I want to keep driving the ESC and the motor without the stall resetting the ESC. I am not concerned about damaging the motor, and the amount of Amps I am drawing should not burn the ESC either.

mathiasvr, 2021-05-17:
Okay, there are some different ways to go about it.

You could change the following part:

https://github.com/bitdump/BLHeli/blob/139f5c6a7a21e1bfa285f9ed166faa1a5fa3d4c5/BLHeli_S%20SiLabs/BLHeli_S.asm#L4529-L4534

Here `4` is the max number of stalls. To disable the stall boost (and avoid it overflowing), you should probably just fix the `Stall_Cnt` like this etc.:

```asm
run_to_wait_for_power_on_brake_done:
	mov	Stall_Cnt, #0 ; Reset Stall_Cnt to disable stall protection and stall boost
	clr	C
	mov	A, Stall_Cnt
	subb	A, #4
	jc	jmp_wait_for_power_on
	jmp	init_no_signal
```

This approach will still make the ESC go back to the startup run phase but not reset completely, not sure if that's okay?

AbhishekJoshi, 2021-05-19:
Thanks for responding @mathiasvr .

> not sure if that's okay?

Ideally, we would like the ESC to keep commanding the motor without any pause/stop. If we could keep the boost up to 100% in the beginning, that would also be ideal.

I'm aware of the risks of doing this.

mathiasvr, 2021-05-19:
> Ideally, we would like the ESC to keep commanding the motor without any pause/stop. If we could keep the boost up to 100% in the beginning, that would also be ideal.

In that case, you might want to remove the following check:
https://github.com/bitdump/BLHeli/blob/139f5c6a7a21e1bfa285f9ed166faa1a5fa3d4c5/BLHeli_S%20SiLabs/BLHeli_S.asm#L2670-L2676

This will make the ESC continue run mode when the comparator feedback times out. I'm not sure how this would work, some re-initialization of the startup phase might be necessary.

For high "boost" I think it should be sufficient to set high startup power, disable rpm protection and command full throttle.

AbhishekJoshi, 2021-05-21:
@mathiasvr 

Thank you so much for your reply. It worked in terms of not re-starting the ESC, what is already awesome. 

Only question is if I still would need to increase a bit more the torque a low rpm/startup, what could I do - even with having already startup power at 1.5, temperature protection off and low rpm power protect off. 

I know I would be risking quite a bit, I know that, but I am very safe on the side of limiting current to the ESC and motor before the ESC. Thank you once again.

mathiasvr, 2021-05-21:
Right, there is actually some more startup protection you can disable.

You can change the following code:

https://github.com/bitdump/BLHeli/blob/139f5c6a7a21e1bfa285f9ed166faa1a5fa3d4c5/BLHeli_S%20SiLabs/BLHeli_S.asm#L1841-L1851

To something like this:

```asm
set_startup_pwm:	
	; Adjust startup power
	mov	Pwm_Limit_Beg, #255				; Set initial pwm limit
	ret
```

I think normally `Pwm_Limit_Beg` will at most be `75` so you can try anything above that to have more startup power. A value of `255` is max and will effectively disable the startup/rampup protection.