bitdump/BLHeli #534: ESC drive LED lights?

pnpapp, 2021-05-18:
Is it possible to modify the ESC program to drive LED lights? I'm planning to use the strip type LED lights on night-flyer, and it would be cool to have control of the lights. The LED has individual control (RGB) but individual or global control is useful. The lights I have control the ground (common power, individual ground).

Would be cool to have independent control of each color, but I think that BL32 has the s.bus control...

Would this interest anyone enought to help?

robomaniac, 2021-08-24:
on modern flight controller you can control LED strips. They have a output for those LED strip. This is much simpler and cheaper to do. You also have a lot of control over how the LED behaves.

No need to use ESC with custom firmware.

google "betaflight led strip"

pnpapp, 2021-08-25:
And if I'm not using a FC?