bitdump/BLHeli #535: BLHeli32 weird stalling, like a stepper Motor

rs-development, 2021-06-09:
I am driving T200 thrusters with a TEKKO32 ESC with BLHeli 32.7 and I am facing an issue where sometimes the Motor does not startup properly. I linked a video with bad settings to reproduce the issue more reliably. What i find interessting ist that even after going off the throttle when I zoom in on the Motor there is still some buzzing sound. The motor has still some increased resistance only after turning it about half a rotation it start spinning more freely again.

I attached my used config and the proposed config from BlueROV for the T200 and a BLHeliS ESC. 

I already tried different Demag and Compensation angles but it only got worse. Any ideas?

My Config for BLHeli32
[BLHeli32_TEKKO32_F3_B - Rev. 32.7 - Multi_210609 - Copy.ixi.txt](https://github.com/bitdump/BLHeli/files/6623844/BLHeli32_TEKKO32_F3_B.-.Rev.32.7.-.Multi_210609.-.Copy.ixi.txt)

Supposed to work with their ESC Design BLHeliS
http://cad.bluerobotics.com/BLHeli_BRBasicESC-R_H_15-Rev.16.6-Multi_170921.ini

https://www.youtube.com/watch?v=5n0rdtZwJuo

sskaug, 2021-06-09:
This may be a case where the friction or resistance is large enough for the field from the stator to "induce" bemf signal without the rotor moving. We have made a dedicated code optimization for such cases, that has previously shown improvement.
A testcode is posted here: https://github.com/bitdump/BLHeli/tree/master/BLHeli_32%20ARM/Thruster%20no%20windup%20testcode

rs-development, 2021-06-16:
Sounds good, I will give it a try when I am back at the lake.

Silvanosky, 2021-09-22:
Hi,

I am having the same issue, would it be possible to have an options for such case in the main firmware instead of testcode ?

Thanks !

sskaug, 2021-09-23:
Testcode now posted.
This mode of usage is not very common for BLHeli_32, so at the moment we do not plan to make it a programmable item.