bitdump/BLHeli #538: 48-96KHz PWM issue

Myllerman, 2021-07-02:
Hello,

I have a Aikon AK32Pro 4-in-1 50A 6S https://www.aikon-electronics.com/index.php?id=38

What i discovered is that in Blheli32 configurator i can only choose 48KHz and 96KHz.
Is there some way of doing 24KHz? 

I saw that you compiled a "half PWM code" to someone with a Tekko before is it possible to do with this ESC?

sskaug, 2021-07-02:
Sure - testcode is now posted.

Myllerman, 2021-07-02:
> Sure - testcode is now posted.

Thank you that was fast! Will test it this weekend. I also contacted Aikon and they would talk to their software guy about 24khz option.

Myllerman, 2021-07-03:
> Sure - testcode is now posted.

It went in without a hitch. Two questions. Now with the new code 48Khz in the settings page is 24khz? To clarify it wont say 24Khz in the interface? Question two, is there any difference between running half speed with custom code vs if the manufacturer had choosen 24-48Khz instead of 48-96khz? Would it run the same?

sskaug, 2021-07-03:
This code runs half of the displayed pwm low frequency. So if the Suite says e.g. 48kHz, then the actual frequency is 24kHz.
And the behavior will be the same as for a code designed for 24kHz.

Myllerman, 2021-07-03:
> This code runs half of the displayed pwm low frequency. So if the Suite says e.g. 48kHz, then the actual frequency is 24kHz.
> And the behavior will be the same as for a code designed for 24kHz.

Thank you for the fast respons and excellent support!

closed 2021-07-03 by Myllerman