bitdump/BLHeli #539: desyncs (or resets)

nablabla, 2021-07-03:
Hi, I am not shure if this is the correct place to post this. I have had a lot of problems latetely.
I switched to 16.9 from the blheli configurator (O-H-5) not shure what exact version that is or where the repository is.  I have two whoops with the happymodel ELRS AOI board, with 0802 motors. First with the 22000kv I experienced sporadic desyncs and/or complete board resets (FC gots stuck) I assume now that it was caused by the ESCs. Reducing ramp up power to 0.5 (from 1.0 default) helped but it still happnes. The other whoop with 25000kv motors does not work at all, even with ramp up power reduced to 0.38
Now I tried the following: I compared 16.7 and 16.9 by going into the motor tab in betaflight, with battery and props on and held it in my hands (its only a whoop after all). Then played with the sliders (making shure the battery does not go below 3.4V)
On 16.7 everyting works, I can go full throttle for some seconds, I can make 20 successive short punches, I can do 60% for a while and so on
On 16.9 48kHz nothing works. If I go above 50% immediately at least one of the ESCs resets. If I hold it at 30% after some seconds some ESCs stops. 
I tried this with both 22000kv and 25000kv motors. I also did the test with JESC (96kHz) and it works as well.
Side node: I also have a 4" with 1404 3500 4S. I had about 15 succesfull flights/batteries with  it then switched to 16.9 and I had two desyncs with 4 flights/batteries, suspiscious...

sskaug, 2021-07-03:
This issue belongs somewhere else. 16.7 was the last official BLHeli_S release.
I believe your 16.9 code is a JazzMaverick release: https://github.com/JazzMaverick/BLHeli/tree/JazzMaverick-patch-1/BLHeli_S%20SiLabs

nablabla, 2021-07-03:
ok, they just do not have an issue tracker? How can I talk to these guys?