bitdump/BLHeli #540: BL_heli temporarily knocks out motor

Ricky-Millar, 2021-07-04:
Hey, having some strange behaviour which im not sure is bl_heli's fault but it seems to be at the moment.

Essentially, if I use bl_heli to even read my ESC, motor 2 on my quad will just stop working. no errors, it can still talk to bl_heli. but it won't spin, in betalight or bl_heli itself

If I unplug and replug the battery then I can happily spin up the motors in betaflight and it works all fine, no errors still. 

It's real strange because I can close bl_heli and the quad is still effected, even though I haven't tried to change any settings on the ESC. I don't understand it at all but I have done it a bunch of times to try and confirm that it is happening and I don't just have a dodgy connection somewhere, but it keeps happening as soon as I open and use bl_heli

Ricky-Millar, 2021-07-04:
https://photos.app.goo.gl/FsRAbmyS8LkAeMyD9

little video of it

4712, 2021-07-04:
@Ricky-Millar It's hard to guess, which BLHeliSuite32 and which Betaflight FC and version you are using. In order to better understand what's going on here, please enable the log function under "Options", repeat the steps you have done before, and post the log.tlg as zip file here. Also it maybe helpful to know, if there is anything special in the ESC settings (beside of reversed #2 and #4).

Ricky-Millar, 2021-07-05:
hey! I tried it on both 3.7 and 3.8 firmware as I was wondering if the update caused it, but it didn't. There were no other special ESC settings!
[Log.zip](https://github.com/bitdump/BLHeli/files/6761464/Log.zip)
here is a log, I just connected and tried to spin up the motors. same behaviour.

4712, 2021-07-05:
@Ricky-Millar thanks for the log. AS far as I can see, everything is just looking normal. Hard to day what causes the problem you are seeing.
When connecting the ESCs, BLHeliSuite32 switches the motor outputs off and configures them for ESC bootloader connection.
When disconnecting, BLHeliSuite32 switches the motor outputs on again. With F7 FC this does in general not work, so there is an option the reboot the FC in BLHeliSuite32. Alternatively one can reboot the FC by dis- and reconnecting the battery.

I could not reproduce the issue with my F4 board here.
Now the question is, why does re-enabling the motor output for ESC2 not work on your FC board.
Maybe there is something special with the hardware on your FC board.
Did you try to disable Bidi Dshot?      
Did you try to update BF to the latest revision 4..2.9?