bitdump/BLHeli #542: Unprecise spin-up with "Non damped mode" = ON

PippoAe, 2021-07-08:
I use Non damped mode to have the motors run cooler on my project. 
It works wonders when the motors are able to just freely spin down but only as the motors are only used in full-throttle.

When not running full-throttle while "Non damped mode"=ON => Very  unprecise RPM responses when spinning up are observed. Sending motor speed of 20% slowly spins the motor up to about 50% over a long period of time.

Other people also noticed this, theres a good video about it here: https://www.youtube.com/watch?v=q36JT1IakZM&t=152s
[image]

I wonder if there is any way to disable motor braking without losing precise RPM response when accelerating or holding throttle.

Thank you very much.

sskaug, 2021-07-09:
Good illustrations you have here. I believe this is just the nature of the electrical/mechanical system.
All the ESC does, is to convert the input signal value to a PWM duty cycle. And that conversion is precise and fast.

PippoAe, 2021-07-10:
@sskaug It results in the motor further speeding up above the set throttle input over a very long time. It's a very sponge like reaction that I'd love get rid off somehow.
Something like no motor braking on the way down, while still reaching and holding higher inputs fast and precise.

Setting 10%  throttle with "non damped mode"=on starts the motor and slowly speeds it up to the RPM the motor normally reaches with 50% throttle while non-damped-mode is off.

PippoAe, 2021-08-06:
> Good illustrations you have here. I believe this is just the nature of the electrical/mechanical system.
> All the ESC does, is to convert the input signal value to a PWM duty cycle. And that conversion is precise and fast.

Dear @sskaug thanks for that response.
So according to your answer everything is fine? Do you see any way to have precise motor spin-up up and throttle hold without braking on deceleration in the current BLHeli_32 version? Non-damped mode is the reason I switched from BLHeli_S to 32, but with this unprecise throttle response it's of no use sadly.

Thanks again.

PippoAe, 2021-09-23:
If this will stay the status quo of "Non damped mode" = ON, I'd like to turn this into a feature request for a new "No brakes on deceleration"-Mode.

BuilderBB, 2021-09-23:
This would be really helpful to alleviate unnecessary heat build up in projects using 3D Printed frames. Please take the brakes off