bitdump/BLHeli #544: How to downgrage to 32.7?

limonspb, 2021-07-21:
Hi

32.8 suite does not show older versions of Blheli_32 to flash, only 32.8
32.7 suite can't connect and flash 32.8 escs.

What is the way to downgrade?
Thanks!

4712, 2021-07-21:
@limonspb If BLHeliSuite32 32.8.x.x does not show 32.7 as option, there is no 32.7 hex file for the specific the ESC available.
It depends on the start date of production. What ESC shows up in BLHeliSuite32?

limonspb, 2021-07-21:
@4712  thanks for the quick answer!
Its Foxeer reaper mini 4IN1 60A esc.
I think they came with 32.7 and i flashed with 32.8. But i could be wrong.
Could you please double-check there are no hex files for them for 32.7?
[image]

4712, 2021-07-21:
@limonspb for FOXEER_Reaper_Mini_4IN1_60A there is no rev. 32.7 or lower hex file, so they must have came flashed with 32.8.
BTW: the latest BLHeliSuite32 is 32.8.0.7.
Is there a special reason, why you wish to downgrade from 32.8?

limonspb, 2021-07-21:
> @limonspb for FOXEER_Reaper_Mini_4IN1_60A there is no rev. 32.7 or lower hex file, so they must have came flashed with 32.8.
> BTW: the latest BLHeliSuite32 is 32.8.0.7.
> Is there a special reason, why you wish to downgrade from 32.8?

Thanks for checking!

Rumors say 32.8 is more noisy especially with variable PWM. So i wanted to check myself, record BBL on the same quad with both 32.7 vs 32.8.

sskaug, 2021-07-22:
We have generated 32.7 code for your ESC here: https://github.com/bitdump/BLHeli/tree/master/BLHeli_32%20ARM/Older%20revs%20testcode

limonspb, 2021-07-23:
@sskaug thanks a lot! How do i flash it?
I mean Blheli suite 32.8 does not have a button to flash local hex files
32.7 does not connect with blheli 32.8 escs.

4712, 2021-07-23:
@limonspb Use either "Flash BLHeli"..."Ignore" and pick the file, or copy the hex file into "\BLHeli32_HexFiles" sub folder and restart BLHeliSuite32.

limonspb, 2021-07-23:
Thanks a lot, guys! Closing the issue. What great support!

closed 2021-07-23 by limonspb