bitdump/BLHeli #547: Missing variable pwm firmware

c0debreaker, 2021-08-06:
I'm not sure where to ask my question since it's not an issue. I'd like to try the patch for variable pwm. I noticed the BLheli devs shared a fix for it. However, I can't find 32.81 from the dropdown BLHeliSuite32 application. I only saw 32.8, 32.7 and 32.6. I also can't find a way to manually load the 32.81 hex file of Spedix GS45A from inside the application. Any help would be greatly appreciated.

4712, 2021-08-06:
@c0debreaker Please use the latest BLHeliSuite32 version. Simply place all hex files you need in the \BLHeli32_HexFiles sub folder and restart the Application.

c0debreaker, 2021-08-07:
Ah cool! I was thinking about that solution. Thanks a lot!

closed 2021-08-07 by c0debreaker