bitdump/BLHeli #549: weird message when I try to connect to escs

SkylionFPV, 2021-08-18:
I get this message when I try to connect to my escs is there a way to fix it?

"found outdated 4way-interface revision v20.0.0.4 

please update "unknown" firmware to the most recent revision, (4way-interface v20.0.0.5 or higher)"

4712, 2021-08-18:
@SkylionFPV the message occurs when there is no ESC found and the 4way-interface part in the flight controller firmware is possibly not able to handle newer MCU types on the ESC boards.
With BF firmware it is highly recommended to use rev. 4.2.7 or later.
If you are using ESC with more classic MCU types like STM32F051, the reason for connection failure is to be search at other places.
Perhaps, you can post a log.tlg file here...
What kind of firmware is flashed on the flight controller?

SkylionFPV, 2021-08-18:
I have the newest version of emuflight (0.3.7-2000) flashed onto the flight controller (link https://github.com/emuflight/dev-unstable/releases/tag/hex-2000), the flight controller is the betafpv f405 v4 20a (https://betafpv.com/collections/brushless-flight-controller/products/toothpick-f405-2-4s-aio-brushless-flight-controller-20a-blheli_32-v4), my confusion is that I have the hglrc zeus13 with blheli32 escs also on the same version of emuflight and I can connect to the escs just fine, the only difference between them afaik is that the zeus used normal f0 mcus and the betafpv used at32f415 instead, could the f415a be causing the problem?

4712, 2021-08-18:
Yes, AT32F415 is one of the newer MCU, which needs 4way-interface v20.0.0.5 or higher to be detected.

SkylionFPV, 2021-08-18:
so should I flash it to betaflight to connect to the esc? is there a way to update emuflight to work I can talk to the emu devs about it if I know what to tell them lol

4712, 2021-08-18:
@SkylionFPV Yes, one option is to flash BF until update of EmuFlight is merged.
Update for EmuFlight was easy: https://github.com/emuflight/EmuFlight/pull/675

SkylionFPV, 2021-08-18:
tested and working with emuflight now thanks for your help @4712