bitdump/BLHeli #550: [BLHeli_32] Applied Startup Sound crashes halfway (-> no communication to FC possible)

bennodev19, 2021-08-20:
Hello there,

Last weekend, I built a Cinewhoop and everything worked out fine.
As `AIO FC` I chose the **'[BetaFPV Toothpick F4 2-4S AIO FC 20A V4](https://n-factory.de/BetaFPV-Toothpick-F4-2-4S-AIO-FC-20A-V4)'** which has an inbuilt `20A BLHeli_32 ESC`.

Since I decided to go with the plug motors,
I had to reverse the spinning direction of two of them in **BLHeli_32**.
After I've successfully reversed the motors (worked fine 👍),
I noticed that I can change the startup sound to a custom sound via **BLHeli_32**
and so I updated it based on these two videos:
https://www.youtube.com/watch?v=4wCmhiaBkjA
https://www.youtube.com/watch?v=_8VD5BeEb8U

## 😬 Problem
However, after I've successfully applied the music to the ESC via pressing 'Apply Music' and 'Write Setup',
disconnecting the ESCs via pressing 'Disconnect' (for testing the new startup sound),
I noticed that the startup sound crashes halfway through (although **BLHeli_32** has successfully applied the music to all `4 ESCs`). And **after the startup sound has crashed, the FC can't communicate with the ESCs anymore** (see video)
https://youtu.be/-lG82S6S1tQ

## 👉 Feature Request
As I've probably failed to copy and paste the correct notes (otherwise it wouldn't crash halfway),
it would be nice if **BLHeli_32** would automatically check if all pasted notes are valid (like the 'Analyze Script' button does) before applying it to the ESCs.
Otherwise, if you aren't careful, you'll end up in the same position as I'm with an AIO FC whose FC can't communicate with the ESCs. (which is bad as a multicopter without working ESCs will never fly ^^)

Thanks a lot ;D

4712, 2021-08-20:
Sorry to read about this failure.
Because the bootloader has no chance to interact with BLHeliSuite32 before the start up music is played till the end, it will be hard to reflash  the ESCs.
We never had that before, and it's hard to guess, what has happened here.
  
Actually there is a build in feature like you requested: On "Apply Music" a Analyze Script function is executed for each ESC channel. But maybe there is something important skipped here.

I will try to get me one of the AIO and dig into the scenario, but that will take a while.

Do you have any idea about what has gone wrong when copying the script? What portion of
`## Motor 1 ## 
Set Music On
Set Gen. Length 15
Set Gen. Interval 0    
Paste these notes:
F5 4 P4 G#5 4 P8 F5 8 P8 F5 8 A#5 4 F5 4 D#5 4 F5 4 P4 C6 4 P 8 F5 8 P8 F5 8 C#6 4 C6 4 G#5 4 F5 4 C6 4 F6 4 F5 8 D#5 8 P8 D#5 8 C5 4 G5 4 F5 1`
did you select for coping?
Did you try to disconnect the motors and retry a connect?

bennodev19, 2021-08-20:
hi 4712,

Thanks for the fast reply ;D

> Actually there is a build in feature like you requested: On "Apply Music" a Analyze Script function is executed for each ESC channel. But maybe there is something important skipped here.

Makes totally sense that the program checks whether the user input is valid, before applying it to the ESCs.
I just wasn't sure, as the track seems to break off in the middle,
which, at the first glance, looks like to be caused by damaged note inputs.

> Do you have any idea about what has gone wrong when copying the script?

To be honestest, I'm relatively sure that I've copied and pasted everything correctly.
My music configuration steps for each single `ESC` were:
1. Went to the YouTube description and copied the notes below `'Paste these notes'` from the corresponding ESC
   -> `ESC 1` got the notes of the YouTube description at `### Motor 1 ###`
   -> something like: `F5 4 P4 G#5 4 P8 F5 8 P8 F5 8 A#5 4 F5 4 D#5 4 F5 4 P4 C6 4 P 8 F5 8 P8 F5 8 C#6 4 C6 4 G#5 4 F5 4 C6 4 F6 4 F5 8 D#5 8 P8 D#5 8 C5 4 G5 4 F5 1`
2. Clicked the 'Music On' checkbox (-> allowed music) (`BLHeli_32`)
3. Pasted the copied notes into the input field (`BLHeli_32`)
4. Changed the 'Gen. Length' to `15` (`BLHeli_32`)
5. Tested the notes via 'Analyze Script' Button (but not sure if I've done that for all ESCs) (`BLHeli_32`)

> Did you try to disconnect the motors and retry a connect?

Yeah, I've tried to reconnect the ESCs to `BLHeli_32` more than once, but the FC can't talk to the ESCs (-> no luck).
I'm currently trying to figure out how to flash the ESCs without going over the FC on an AIO FC.
(If you have an idea let me know ^^)

> Did you write the same or different music to each of the 4 ESC?

`ESC 2` and `ESC 3` have the same music notes but `ESC 1` and `ESC 4` have different music notes.

Luckily, I've the same Board lying around (with a damaged OSD (factory issue), 
that's why I got another one (this one), which I killed by flashing the music ^^) .
As a FC without OSD isn't useful anyway, I can try to follow the same steps I've followed before
and see whether the starting sound crashes there too.

Thanks a lot ;D

bennodev19, 2021-08-21:
Unfortunately, it looks like to be an issue with `BLHeli_32` (or bad luck twice) 
as the same issue happens to the same AIO Board (not the same but the same model, brand, ..) again.
This time I've documented everything and saved the exact logs. 
So that it's easier for you to comprehend this issue.
Hopefully you can find what it is causing,
because otherwise other people might run into it and kill (idk if its killed, but I couldn't recover it yet) their AIO FC.

Logs Image:
[image]

Logs File:
https://drive.google.com/file/d/1KJHHPr2aDImwge_OrYOxKtwXmRvjfx6F/view?usp=sharing

Video:
(Coming soon, have to edit it as I've recorded it from 3 different angles)

Thanks a lot ;D

sskaug, 2021-08-21:
Thanks a lot for reporting so thoroughly!
That makes it easy to dig into what is happening here.
And it turns out that it is a result of the very high clock frequency (148MHz) of the AT32F415 MCU.

What happens is that a 16 bit timer is compared to a number that is more than 16bits:
[image]

Unfortunately I do not at the moment see any way of recovering dead ESCs.

There are two actions we should take here.
Of course such events must be prevented in future firmware releases.
And in the meantime we should implement a filter in BLHeliSuite32 to prohibit writing such data.

bennodev19, 2021-08-22:
Glad you could find the problem so quickly 👍
However, I'll keep my hands off the music feature for my next AIO FC for now ^^

> Unfortunately, I don't see any way to restore dead ESCs at the moment.

Do you think the ESCs are dead after the startup sound crashed, or just lose connection to the FC.
Is looking for a way to connect to the ESCs without the FC pass-through pointless anyway?
Since you know (better than I do) what went wrong, 
it would be cool if you could give me your estimation ^^

Thanks a lot ;D

> There are two actions we should take here.

Really good of you guys to support 
and respond so quickly to the problems of the community. 
Keep up the good work 👍

sskaug, 2021-08-22:
Thank you for your understanding here. Unfortunately the code gets stuck in the midst of music, before it reaches bootloader check and normal operation. So it will not operate normally either.

benedikt-bartscher, 2021-08-22:
@bennodev19, maybe you can get an ST-Link and flash AM32 (https://github.com/AlkaMotors/AM32-MultiRotor-ESC-firmware). I dont know if your ESC is supported yet, but you could ask in their slack channel!

sskaug, 2021-08-23:
I'm not sure AM32 supports this M4 MCU yet.
Of course we will compensate your loss, email us at blheli32 at gmail dot com

Uvamosk, 2021-12-28:
I made a account to just post here. I have the exact same issue. Took a brand new BetaFPV F405 20a board used the configuration and changed motor directions and flashed and worked fine. Tested motors in BetaFlight. I then made the decision to do the music flash and had the same issue the OP had. I flashed the music everything said success. I then power cycled heard a single chirp then nothing. 
 I also have all the same issues OP has. I can connect to USB connect to the ESC's but then reading from them gives me the exact same response that is in the OP's Video... I feel your pain OP. I bought another board and pretty much fear touching the music. Also very iffy about doing any flashing at all now. Never worried about it on the Blheli_s stuff.

sskaug, 2021-12-28:
Sorry to hear about your loss and fuzz. Please email us at blheli32 at gmail dot com.
You must have used an older version of BLHeliSuite32, a fix has been implemented for this in newer versions.
It is also fixed in the Rev32.8.3 testcode.