bitdump/BLHeli #552: 32.7 hex request for Foxeer Reaper 96pwm version

d00m303, 2021-08-29:
I have foxeer reaper 65A 96K esc, it comes with 32.8 blheli version. Im having a lot of desyncs and i need to check is it related to blheli 32.8. I also checked 32.8.1 and quad still desyncs on sharp moves. 
Can i request 32.7 target for this esc? I see on old targets just reaper non 96k version to download.
Many thanks! ;)

sskaug, 2021-08-30:
Now posted. Note that 32.7 with STM32F051 does not run well on higher pwm frequencies.

pdxtigre, 2021-08-30:
Having HOLYBRO TEKKO32 F3 35A ESC BLHELI_32 on a 250 frame, iFlight XING-E 2306 2450KV with both 32.7 and 32.8 horrible experience as d00m303.

d00m303, 2021-08-30:
> Now posted. Note that 32.7 with STM32F051 does not run well on higher pwm frequencies.

Im using 48/48. If You can share 32.7 for this esc just to troubleshoot, thanks ;)

sskaug, 2021-08-31:
Sorry, created the code but forgot to push. Now it is there.

d00m303, 2021-09-01:
> Sorry, created the code but forgot to push. Now it is there.

Hey Sskaug, that totally fixed my issues. Many thanks!

closed 2021-09-01 by d00m303

Benxw, 2022-05-31:
Hello, where i can find the 32.7 version for the esc Foxeer Reaper ? Thank you