bitdump/BLHeli #553: rev 32.7 not available

varefaz, 2021-08-30:
with version 32.8 the esc are not always recognized and often do not arm the motors. I would like to put the version 32.7 for the stack rush blade f7 analog but it is not in the firmware choice menu.

sskaug, 2021-08-31:
Please let us know the exact name of the BLHeli_32 firmware, as shown in BLHeliSuite32

varefaz, 2021-09-01:
> Please let us know the exact name of the BLHeli_32 firmware, as shown in BLHeliSuite32

RUSH_BLADE_SPORT_V2

sskaug, 2021-09-01:
Unfortunately the MCU used in the RUSH_BLADE_SPORT_V2 is not supported in earlier revs than 32.8.

frat1234, 2021-09-01:
> Unfortunately the MCU used in the RUSH_BLADE_SPORT_V2 is not supported in earlier revs than 32.8.

I have the same problem. Please i need to return to rev 32.7 to solve this problem. Can you send us the download of 32.7 for RUSH_BLADE_SPORT_V2?

sskaug, 2021-09-01:
Sorry, no, as this MCU is not supported in earlier revs than 32.8.
You should contact your vendor regarding these problems.

varefaz, 2021-09-01:
> Sorry, no, as this MCU is not supported in earlier revs than 32.8.
> You should contact your vendor regarding these problems.

Can you pls build a 32.7 version for this esc as you have done for another guy (he has a foxeer esc)? we need it.I have two other friends who have the same problem