bitdump/BLHeli #555: cmnd_stk_get_sync

PaulosKapa, 2021-09-06:
i have connected my esc (xxd hw30a) to my arduino uno and i want to flash it so it can go in reverse too. When i go to make interfaces, choose the board, click on arduino blheli bootloader and then upload the sketch to my arduino it fails and it pops up this msg (cmnd_stk_get_sync). I can upload sketches from arduino ide to my arduino and the esc works. What is going on?

4712, 2021-09-06:
cmnd_stk_get_sync is general answer for connection to Arduino bootloader failed. This might happen, if the com port is chosen wrong, the Arduino type is chosen wrong or the baudrate is wrong or by other reason. Some old Uno bootloader use 57600 baud, maybe useful to try.

PaulosKapa, 2021-09-12:
> cmnd_stk_get_sync is general answer for connection to Arduino bootloader failed. This might happen, if the com port is chosen wrong, the Arduino type is chosen wrong or the baudrate is wrong or by other reason. Some old Uno bootloader use 57600 baud, maybe useful to try.

had the correct com port, tried different usb ports, different baudrates, had the option uno w/ atmega328 chosen, still the same problem. If instead of clicking on make interfaces i click on read setup it shows this message: connection to 4-way interface failed please check usb/serial port -and power- connections to the interface

4712, 2021-09-12:
@PaulosKapa what is actually the version of BLHeliSuite you are using?

PaulosKapa, 2021-09-13:
@4712   BlHeliSuite32_32811, windows 10 64 bit. I ll try with an arduino nano too.

PaulosKapa, 2021-09-13:
@4712 i tried it with a nano, i get the same results. The arduino ide uploads the sketch normally.

4712, 2021-09-15:
@PaulosKapa never had this issue before. BTW for 8-but ESC you need to use BLHeliSuite, not BLHeliSuite32. But as far as I can see from the [schematic](https://github.com/NicksonYap/XXD-HW30A-ESC-Schematic/blob/master/images/schematic.png), you will not be able to use BLHeli with xxd hw30a, it has no external XTAL or ceramic resonator.

jvfd3, 2023-07-25:
@4712 I had a similar problem with and Arduino UNO and solved it by setting Baud to 115200 and selecting the option Arduino BLHeli Bootloader

JiaShengJerryLu, 2024-01-28:
Had this exact issue, I ended up resetting my arduino's eeprom (see "Resert EEPROM Memory" section at this link: https://support.arduino.cc/hc/en-us/articles/5779192727068-Reset-your-board) and the issue was resolved!