bitdump/BLHeli #558: FrSky Neuron 40S telemetry

shellixyz, 2021-09-08:
Hello,

I'm trying to get Kiss style telemetry working instead of S.Port telemetry on the FrSky Neuron 40S but it doesn't output anything on the telemetry pin when the S.Port physical ID setting is set to off. I'm using DSHOT and the FC is sending telemetry requests. I even tried with telemetry auto but still nothing. I scoped the output and it is just a flat line.

Could it be that the firmware for this ESC has "normal" / non-SPort telemetry disabled ?

sskaug, 2021-09-09:
The FrSky Neuron ESCs are very special. Before BLHeli_32 implemented S.PORT, FrSky implemented it with their ESCs in a dedicated, separate MCU. SO in the Neuron ESCs, there is a MCU that converts the KISS style telemetry to S.PORT telemetry.

shellixyz, 2021-09-09:
Ah, makes sense, otherwise even if BLH32 had SPort when they designed the ESC they couldn't implement the SBEC configuration through SPort without changes to BLH32. I guess I should be able to tap into the telemetry line directly then. Thank you

closed 2021-09-09 by shellixyz