bitdump/BLHeli #559: SpeedyBee 4 in 1 ESC RPM FILTER warning

abimaputra78, 2021-09-10:
Dear BLHELI32 Team,

I used Speedybee F7 and had my new ESC, it's blheli32 Speedybee 4 in 1.
I had setup everythings proper, update your latest version, and bf 4.2.9 which was setup 8k/8k dshot 600. All the things follow what bf 4.2 tuning notes.

But I had issue that still unsolved till now, when I plugged li po, there was a warning OSD message  "RPM FILTER", I have to reconnected LIPO and again and again to make its warning gone., but sometimes there was no warning, sometimes there was. It looked so funny and annoying. 

I need your help for this solutions, since it is related with safety and I dont want something bad happen in the air. 🙏 

I really appreciate your help and confidence, if this problem solved Sirs. 

Thanks
Ditto.

sskaug, 2021-09-11:
This is most likely not directly related to BLHeli_32. At least it should be investigated by the Betaflight team first.

abimaputra78, 2021-09-11:
Hi Thanks for your email,

I did Sir, I had read some of their issues here.

https://github.com/betaflight/betaflight/issues/9840

Thanks a lots. 🙏

sskaug, 2021-09-11:
Ok, so that particular fix was in Rev32.8.

abimaputra78, 2021-09-11:
Hi,

Thanks so much. 🙏 Really appreciate.

Thanks.

abimaputra78, 2021-09-11:
I had 32.8.1.
But this issue still there. Should I try 32.8.2? But it seemed not release
yet

AlanzFPV123, 2021-10-30:
I have a similar issue with this stack. 2 related problems- I think it’s an esc hardware issue but not an expert. 
1) motors do not all respond to throttle upon arming every time. 
2) Blackbox logs cannot be parsed. 

More detail: If all motors do respond on first arm after powering quad, then BB log is readable, as long as I power down and download log (or view via Bluetooth on SpeedyBee app)
If however one or more motors do not respond to throttle on ANY arming event on same flash memory (same flight session) then ALL BB logs in memory are corrupted. Corrupted logs return error message in BBE (latest) that logs cannot be parsed. Also the logs are 16,000kb (memory full) even if just one log. 
The motors that do not respond will be different each time I power the quad and sometimes 2 will not respond. Power cycling the quad solves the issue and quad flies great. This is true 100% of the time. Power cycle always results in all motors working, but if there is just one “bad” arm in flash memory then I get 0 readable logs. 
Speedybee customer service has strung me along for almost 3 weeks now and while they respond to emails, they want me to take my brand new build apart, thy their new beta test app etc. They do not listen very well.

AlanzFPV123, 2021-10-30:
Here’s a link to both bad and good logs:
https://drive.google.com/drive/folders/1iDvBjBcKNc2cXLsG4TZ1LyGhePd6ieqe

AlanzFPV123, 2021-10-30:
I worked though some possible issues like reflashed FC (was on 4.2.8, now on 4.2.9) disabled RPM filters, changed from Dshot to Multishot with LordQWERTY on UAV Tech discord and made no progress. Behavior remains same.

abimaputra78, 2021-10-30:
I just curious i deleted betaflight and change to Inav. All were default
but the issues seem gone but the blheli last beep were not in uniform beep.
I had a bad feelin about this. But the flight performance not as good as
betaflight. I change to bf again sometimes I had issues again. Dont have
any idea.

AlanzFPV123, 2021-10-30:
Interesting. So back on BF but sometimes motors don’t respond?

abimaputra78, 2021-10-30:
Yes Sirs exactly,
Not all the motors but randomized. Just one or two them. Restart of Fc will
solve the problem a while.

In Inav the last to important "beeps" is not uniform. One was leading the
others sometimes but often happen. 🙏

Hope this will have some solution.
Thanks. 👍

AlanzFPV123, 2021-11-02:
Thanks for the reply and feedback. SpeedyBee jut wrote me back and say they will replace the stack and want me to return mine to them. It took 3 weeks to get that from them, and now the added time for shipping both ways etc but at least they seem to be doing the right thing. Hope you have sucess also.