bitdump/BLHeli #560: Only 1 motor not working when variable PWM is configured as 48KHz - 48KHz

FasaArg, 2021-09-13:
Sorry to bother you with this. And i offer my apologies in advance if i am doing something wrong.
I have an old rig for today's standards, but it always worked flawlessly for my needs and i'm sticking with it.
The quad have a Kakute F7 AIO and 4 Aikon AK32 35A ESCs.
It was flying perfectly with sine mode and 48KHz PWM with BF 3.5.1 and an older version of BlHeli (around 32.5 or 32.6)
Now i'm trying with Betaflight 4.2.9 and BlHeli 32.8
All motors works OK on defaults BlHeli values, also with sine mode, as in 24 KHz - 48KHz, even if is configured between with 47 KHz and 48Khz.
But... as soon as is set in 48 KHz - 48KHz, the motor 2 does not start, or it starts randomly some times (less than 1 in 20 times that i plugged the battery)

Facts:

- This affects only the motor 2 (none of the others fail)
- If it starts (without changing anything anywhere) it works ok as always.
- This is manifested with the same results from the motor tab in blhelisuite, the motors tab in Betaflight configurator or arming the quad with the radio.
- I tried flashing blheli 32.7 and it does the same.
- It does not matter if the board is initialized via USB first and then the ESCs are polarized with the battery or if the the battery is plugged directly to init both at the same time.
- The telemetry always works and shows 0% errors in all 4 ESCs (in betaflight and blheli suite)
- I have tried both Dshot 300 and 600.
- I have even tried with 3s and 4s batterys (don´t ask me why...)
- I have run out of ideas to try.

Attached is a DUMP and an overview of the configuration in BLHELI SUITE

[image]
[BTFL_cli_CHAM_TI_20210913_122333.txt](https://github.com/bitdump/BLHeli/files/7155937/BTFL_cli_CHAM_TI_20210913_122333.txt)

Thanks in advance.
Sorry about my poor English.

sskaug, 2021-09-14:
Are you sure this is not a motor or ESC hardware problem?
Have you tried swapping motors and ESCs?

FasaArg, 2021-09-14:
Hi Stephen! How are you? 
No, i didn't tried that because the quad is working since day one, also, i have not spare parts anymore.
I have assumed that if it worked in every configuration except this, the problem could be software related.
My hope was that perhaps this problem have occurred before and some one could guide me to a configuration related solution.
Since this scenario does not ring you any bells and i can not do any further testing due to my lack of spares, I will proceed to close this issue and use it below 48Khz in the low PWM frequency.
Thanks a lot.
Regards.

closed 2021-09-14 by FasaArg