bitdump/BLHeli #561: Dshot specifications and DIGITAL_CMD_BEEP1

theely, 2021-10-06:
Hi,
I am working on a Dshot ATMega4809 microcontroller implementation to drive BLHeli 32 and BLHeli_S ESCs.
So far I managed to successfully set throttle values for BLHeli 32 ESCs at 150, 300 and 600 Kbps.
Next I would like to control motor buzzing using the DIGITAL_CMD_BEEP1 command but I had no luck.

Where can I find specifications on how to use the DIGITAL_CMD_BEEP1 command?

What I have found so far is:
-motors should be idle, I suppose this means throttle value 48 should be sent before issuing the DIGITAL_CMD_BEEP1 command.
-After issuing command DIGITAL_CMD_BEEP1 ensure a minimum daly of 260us before sending any new command.

Is that all? Something else I could try?

mathiasvr, 2021-10-06:
Yes motors need to be completely stopped before sending commands, you can also send a value of 0 for this.

theely, 2021-10-06:
Thanks.
I made sure the motors were completely stopped and used value 0 but I am still not able to trigger the beep.
If I connect the ESC to a BF flight controller I can make the ESC beet so this excludes any issue with the ESC.

Do you know if the DIGITAL_CMD_BEEP1 needs to be sent multiple times before adding the delay. Or it should be sufficient to send it one?
What I am currently trying is to continuously send DIGITAL_CMD_BEEP1 with an interval of 260us between the commands.

mathiasvr, 2021-10-12:
@theely you increase temp4 for beep length and temp3 for pitch.