bitdump/BLHeli #567: No Mapping for the Unicode character exists in the target multi-byte code page

jonasbouchraiet, 2021-11-12:
[image]

I get this Message over and over again. It is a Tekko32 F4 ESC flashed with 32.8 from factory. All the motors spin fine when raising slider in Betaflight motor's tab.

jonasbouchraiet, 2021-11-12:
I only get this message when Motor 4 is connected to the Flight controller. But all Motors turn just fine, when I spin them up. + I get no second sets of beeps on Motor 4, but it still spins up fine. That's the weird part

jonasbouchraiet, 2021-11-12:
I do have a ST-Link, if that's gonna help

4712, 2021-11-12:
@jonasbouchraiet For sure it's an issue of the BLHeliSuite32xm app...
when does the error message exactly occur? Always if you "Read Setup", or even if start the app and wait?
What is exactly your OS version?

jonasbouchraiet, 2021-11-12:
I made a little video: https://www.youtube.com/watch?v=2-PTUOss39I

I get no beeps from Motor Number 4. No startup beep and no signal beep. But the motor spins up fine.

I don't think it is a problem with the App or my OS version. My OS version is: 11.6 Big Sur

4712, 2021-11-12:
Thanks for the video! 
Please try "?... Check for BLHeliSuite32xm update" and see if there is any error?
Then, please turn on the log function "Options... Enable Log", produce the error, then switch to log tab and save log to file (Ctrl + Mousebutton) and post here as zip.
I suspect, that there is a memory corruption in the parameter section and the app cannot read the content, because of character mapping issues.

jonasbouchraiet, 2021-11-12:
When I hit the check button I get the exact same error

jonasbouchraiet, 2021-11-12:
[Log.zip](https://github.com/bitdump/BLHeli/files/7529568/Log.zip)
Here is the log

4712, 2021-11-12:
Thanks, I will look at it.

4712, 2021-11-12:
I will get back soon with a version of the BLHeliSuite32xm , which let you store the parameter section to a file. This will hopefully help to analyze the issue.

4712, 2021-11-12:
@jonasbouchraiet Please download [this beta version](https://app.box.com/s/vc57q44jz3nbi39ofqm9okzpj6aoggfj) and try to read  ESC4. This beta version will hopefully create a file in the documents folder named ESC4.txt. Please post it here as zip.
Thank you!

jonasbouchraiet, 2021-11-13:
Here is the file: 
[ESC4.txt.zip](https://github.com/bitdump/BLHeli/files/7531340/ESC4.txt.zip)

Hopefully you can identify it!

4712, 2021-11-13:
@jonasbouchraiet Thanks a lot, very helpful!!! Now, I can work on it and get back with a fix.

4712, 2021-11-13:
@jonasbouchraiet Please download [a beta with a fix here and retry.](https://app.box.com/s/2bjik5ztk7f8ddjb3dgfxxzn1wh13y7r)
There happened a memory corruption when the bootloader restored the settings area, maybe due to power loss. So there where unknown characters in the ESC name field. The macos version reacted with the encoding exception. 
Now it should read the ESC as unknown, invite to erase the settings area and reflash.
Thanks for your patience and for bringing this up here!

jonasbouchraiet, 2021-11-13:
https://www.youtube.com/watch?v=1FD5EeqgzHk

Problem solved. I can't thank you enough!

closed 2021-11-13 by jonasbouchraiet

mrh666, 2021-12-05:
I'm experiencing absolutely the same issue with 1.0.3.2 BLHeli Suite and Tekko 32 f4 4-in-one ESC with default factory setting. 
How to fix it guys? Where to get beta version (link is not working anymore)?

4712, 2021-12-05:
@mrh666 sorry, link is active again now.

mrh666, 2021-12-05:
I just installed this beta and I'm not sure that new ESC should be recognized like this: [image]
And ESC fully functional from Betaflight motors or armed by Radio:
[image]
Any idea how can I fix it?

4712, 2021-12-05:
@mrh666 It might be the result of too fast production process. Please reflash ESC1 and 4.